ms-iot/vscode-ros #48: Incompatible Python environment could be auto-activated in ROS terminal

seanyen, 2019-05-23:
I have `ms-python.python` extension installed and it by default will automatically launch a Python environment for every terminal open in VSCode, and which results in unexpected behaviors. Any thought?

[image]

kejxu, 2019-05-23:
I am not seeing this behavior on my machine (Windows), let me try it on Ubuntu

seanyen, 2019-05-23:
I think it is to do with this feature: https://code.visualstudio.com/docs/python/environments#_select-and-activate-an-environment

I added `"python.terminal.activateEnvironment": false` in my workspace `settings.json` and then this behavior is gone.

kejxu, 2019-05-24:
Let me do a bit more testing on this and maybe we should add the `"python.terminal.activateEnvironment": false` when the extension gets activated.

meantime, could you help check which Python interpreter you selected for the workspace?
[image]
is the python2.7 from c:\opt\python27amd64 selected?

seanyen, 2019-05-24:
[image]

No, a virtualenv is chosen instead. (My computer has several Python installations.)

kejxu, 2019-05-24:
> No, a virtualenv is chosen instead.

I think this might be the reason, as the Python extension detects that the interpreter is from venv, it will attempt to activate the environment in the terminal

> And here are the choices I have. Do I need to source `c:\opt\ros\melodic\setup.bat` before running `vscode`?

Yes, I think so. our (ros) python install path needs to be in the environment so that the extension can pick it up. Maybe you could also try typing that in manually, I think that would also work.

I am guessing it's because I have gone through this earlier, so when I launch a vscode instance without sourcing `setup.bat` first, I can still see the (ros) python:
[image]

seanyen, 2019-05-24:
Sourcing `c:\opt\ros\melodic\x64\setup.bat` does the trick to make `c:\opt\python27amd64\python.exe` on the list. And I need to manually switch to it once to make the change happen in workspace `settings.json`

kejxu, 2019-05-24:
> Sourcing c:\opt\ros\melodic\x64\setup.bat does the trick to make c:\opt\python27amd64\python.exe on the list. And I need to manually switch to it once to make the change happen in workspace settings.json

this is a great point! We can totally configure this in our extension, I'll test it out and send out a pull request with the change later