ms-iot/vscode-ros #50: [feature] add docs how to install this addon from source?

andrejpan, 2019-05-24:
I am new in VS Code and trying to set up the ROS plugin. As far as I see there are multiple ways. 

First, I can install an old extension through Extension Marketplace from ajshort, but this project is not actively developed any more. I am not managing to build my packages...

Second, apperantly I could install from VSIX - [link](https://code.visualstudio.com/docs/editor/extension-gallery#_install-from-a-vsix). I searched for *.vsix here, but I did not found any suitable file. Is this repository and *.vsix have some intersection or there is another way how I could install this extension?

Best, Andrej

kejxu, 2019-05-28:
hi @andrejpan,

Thanks for the great question! This is a great point, our apologies for the confusion!

> First, I can install an old extension through Extension Marketplace from ajshort, but this project is not actively developed any more. I am not managing to build my packages...

We are pushing towards releasing this extension to the Extension Marketplace and after that you'd be able to find it there =)

> Second, apperantly I could install from VSIX - link. I searched for *.vsix here, but I did not found any suitable file. Is this repository and *.vsix have some intersection or there is another way how I could install this extension?

The easiest way to do this (at this point) is to download VSIX from the build artifact:
1. click on the build status badge
[image]
2. switch to the "Summary" tab and you could download the VSIX from the artifacts
[image]

Let us know if this works =)

Thanks,
James

andrejpan, 2019-06-03:
Hey @kejxu,

thank you for this help. I managed to install this plugin, now I just need to learn how to use it ;-)

During the installation I got a warning:
```
Installing extensions...
(node:13530) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
Extension 'vscode-ros-dev.vsix' was successfully installed.
```
Somebody might need to open new issue?

Best, Andrej

kejxu, 2019-06-03:
thanks for trying it out @andrejpan! Please feel free to reach out if you find any problem/question while using it, always welcome =)

The warning looks like a bug to me (thanks for posting it here!), but it somehow looks like a bug related to Node.js, so it might be a bug in vscode (this is just my guess, I need to check on that). I think it would be great if we could log it as a bug anyhow; would you be able to create a bug and log dependency (only vscode should be needed) version?

Thanks a lot!
James

andrejpan, 2019-06-04:
I am not sure if I understand correctly your last sentence... I should create a new issue and describe with all the necessary information?

Best Andrej

kejxu, 2019-06-04:
> I should create a new issue and describe with all the necessary information?

My apologies for the confusion! Yes, please =) This seems like a separate issue.

It would be great if we could have vscode's version, the steps you took to install the vsix, and anything you'd consider necessary included in the bug report =)

linknum23, 2019-10-16:
I had similar issues and followed some documentation on the vscode-go wiki. I adapted it to BUILDING.md in #137 in the hope that this will help future testers and contributers.

closed 2019-10-22 by seanyen