ms-iot/vscode-ros #56: [bug] Could not source the workspace space setup file.

andrejpan, 2019-06-03:
**general information**
\<Ubuntu 16.04, ROS Kinetic\>
\<Version of the VS Code 1.34.0>
\<Version of the plugin: 0.4.2\>

**describe the bug (*actual behavior*)**
When I start VS Code I get message: `Could not source the workspace space setup file.` Not sure if this is a bug, but I have no idea how to solve it.

**steps to reproduce**
1. source system and local ros enviroment in bash
2. run code in the same bash window

**expected behavior**
No warning/error should be seen.

**screenshots**
[image]

**additional context**
\<any additional information could go here\>

kejxu, 2019-06-03:
nice catch, this should be one of the testing scenarios, sending out fix soon

closed 2019-06-03 by seanyen

kejxu, 2019-06-03:
hi @andrejpan, `shell: "cmd"` was added when we just started porting the extension, this Windows-only specification has been removed in the pull request referenced above to fix for Linux.

Due to the limitation of installing from .vsix, you might need to uninstall and reinstall with the new .vsix published from the build pipeline.

andrejpan, 2019-06-04:
I can confirm that notification disappeared with the updated version of a plugin.

kejxu, 2019-06-04:
> I can confirm that notification disappeared with the updated version of a plugin.

This is super helpful! Thanks for trying it out and reporting this issue! Please feel free to reach out again whenever there is any other issue/question/feature request!