ms-iot/vscode-ros #72: [bug] cpp intellisense not working

kejxu, 2019-07-12:
**general information**
Windows
0.4.5

**describe the bug (*actual behavior*)**
intellisense not working due to import error

**additional context**
includePath added to c_cpp_properties need to be checked, path like `/usr/include` breaks intellisense

Brabalawuka, 2019-07-15:
I am still getting intellisense error after updating the preperties json file

`include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (C:\test\src\helloworld\src\listener.cpp).
cannot open source file "boost/math/special_functions/round.hpp" (dependency of "ros/ros.h")`

My json file is like:  

`{
    "configurations": [
        {
            "browse": {
                "databaseFilename": "",
                "limitSymbolsToIncludedHeaders": true
            },
            "includePath": [
                "C:/opt/ros/melodic/x64\\include",
                "C:/opt/rosdeps/x64\\include",
                "${workspaceFolder}/**"
            ],
            "name": "Windows",
            "intelliSenseMode": "msvc-x64",
            "compilerPath": "C:/Program Files (x86)/Microsoft Visual Studio/2019/Preview/VC/Tools/MSVC/14.22.27812/bin/Hostx64/x64/cl.exe",
            "windowsSdkVersion": "10.0.18362.0",
            "cStandard": "c11",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
}`

kejxu, 2019-07-15:
> I am still getting intellisense error after updating the preperties json file
>
> `include errors detected. Please update your includePath. Squiggles are disabled for this translation unit [...]

thanks for sharing! let's use https://github.com/ms-iot/vscode-ros/pull/73 for this conversation =)

kejxu, 2019-07-16:
@Brabalawuka, thanks for sharing your experience and results with us! I have merged https://github.com/ms-iot/vscode-ros/pull/73, and a newer version of the extension can be downloaded and installed via [vsix](https://ros-win.visualstudio.com/ros-win/_build/results?buildId=4669&view=artifacts).

Would love to know if this works for you =)

Meantime, we'll close this issue for now. Please let us know if this needs to be reopened anytime, thanks again!

closed 2019-07-16 by kejxu

Brabalawuka, 2019-07-16:
> @Brabalawuka, thanks for sharing your experience and results with us! I have merged #73, and a newer version of the extension can be downloaded and installed via [...]

Hi, thank for your help! 

The newer version is still getting me the `cannot open source file "sys/time.h"` error
```{
    "configurations": [
        {
            "browse": {
                "databaseFilename": "",
                "limitSymbolsToIncludedHeaders": true
            },
            "includePath": [
                "C:\\test\\devel\\include\\**",
                "C:\\opt\\ros\\melodic\\x64\\include\\**",
                "C:\\opt\\rosdeps\\x64\\include\\**"
            ],
            "name": "ROS",
            "intelliSenseMode": "msvc-x64",
            "compilerPath": "C:/Program Files (x86)/Microsoft Visual Studio/2019/Preview/VC/Tools/MSVC/14.22.27812/bin/Hostx64/x64/cl.exe",
            "cStandard": "c11",
            "cppStandard": "c++17"
        }
    ],
    "version": 4
}```

kejxu, 2019-07-16:
> The newer version is still getting me the `cannot open source file "sys/time.h"` error

As we have discussed, the error about `sys/time.h` likely requires an update of ROS, not related to this extension.