ms-iot/vscode-ros #77: Geeting invalid macro error after generating cmake cache in visual studio

Brabalawuka, 2019-07-17:
**Describe the bug**
E0992	command-line error: invalid macro definition: =

**To Reproduce**
Steps to reproduce the behavior:
Open the workspace and generate cmake cache

[image]

[image]

**Expected behavior**
Getting intellisense and works fine

**Environment information:**
 - OS version: `win10 1903`
 - Chocolatey Versions: `choco list --local-only`

Brabalawuka, 2019-07-18:
This can be fixed by a series of configuration, I have written a detailed tutorial about this configuration:

https://github.com/Brabalawuka/RosOnWindows

In the `CMakeSettings.json` file, do the following:

->>`"generator": "NMake Makefiles"` ->> This because `Catkin_make` uses `NMake` instead of `Ninja`  
    ->>`"configurationType": "RelWithDebInfo"` ->> Match with Catkin Type  
    ->>`"buildRoot": "C:\\catkin_ws\\build"` ->> Change build root to **your own workspace folder** (refer to you solution penal on the right side)  
    ->>`"installRoot": "C:\\catkin_ws\\install"` ->> Change install root to **your own workspace folder** (refer to you solution penal on the right side)  
    ->>`"cmakeCommandArgs": "DCATKIN_DEVEL_PREFIX=C:\\catkin_ws\\devel"` ->> Change develpment root to **your own workspace folder** (refer to you solution penal on the right side)  
    ->>~~`"buildCommandArgs": "-v"`~~  
    ->>`"cmakeExecutable": "C:/opt/rosdeps/x64/bin/cmake.exe" ` ->> Add this line to use ROS default cmake instead of vs one to avoid confusing errors.  

Refer to following pictures:  

[image]  
[image]

These could probably sove those intellisense errors and can successfully compile and run.

ooeygui, 2020-11-19:
Please reopen if this still repros

closed 2020-11-19 by ooeygui