ms-iot/vscode-ros #79: [feature] Debugging and breakpoints support

mElBalkini, 2019-07-29:
**Is this feature request related to a problem?**
One of the most important features of an IDE is the debugging tools. A breakpoint and a stack trace would save few hours of writing print statements.

**What kind of solution would be helpful?**
Similar debugging functionality to any IDE. Clion is a good example, just press the bug button and it simply stops on breakpoints. Sometimes you have to do manual setup of input arguments to the node though!  
It would be great to have support for roslaunch debugging too, however support for rosrun or directly involving the node with the right arguments and environment variables should be good.

rcabg, 2019-08-02:
I join this request. The possibility of debugging from a roslaunch file would be great since code changes wouldn't be necessary for the gdb walkaround.

@mElBalkini while this feature is not available... do you know this? It works great for me [](https://medium.com/@tahsincankose/a-decent-integration-of-vscode-to-ros-4c1d951c982a)

ooeygui, 2019-08-07:
We are working on this. First we're enabling [attach to process](https://github.com/ms-iot/vscode-ros/pull/87), then debug from roslaunch.

kejxu, 2019-08-08:
Thanks for the great suggestion! This is indeed a great point! As Lou has mentioned above, we have checked in #87, the dev version (aside from the released version on marketplace) could be downloaded from [here](https://dev.azure.com/ros-win/bed058dd-46da-4029-bb85-25eae7674d09/_apis/build/builds/4882/artifacts?artifactName=VSIX&api-version=5.2-preview.5&%24format=zip) and installed as `.vsix`. Please try it out and check how it works!

We are still working on enabling debugging from roslaunch, and would love to get all kinds of ideas and insights for the debugging functionality =)

rcabg, 2019-08-09:
@kejxu @ooeygui thanks a lot for the work you are doing! I will check it out for sure. Any instructions of what I have to do once I install the dev version?

kejxu, 2019-08-09:
> Any instructions of what I have to do once I install the dev version?

Great point! We'll make sure to have documentation coverage for the final version =) The easiest way to test out the change would be like this:

1. add a new "ROS: Attach" debug configuration (by clicking on `Add Configuration...`)
[image]

2. choose the newly added configuration and launch (by pressing `F5` or the blue arrow)
[image]

3. choose runtime type and pid in the picker

*P.S. current implementation for C++ debugging uses the `cppvsdbg` debug type from `vscode-cpptools`, which is only for Windows; meantime, C++ debugging needs Visual Studio environment properly sourced in the VS Code instance (VS Code has to be launched from a command prompt with VS dev environment sourced). This change (C++ and Python) has not been fully tested on Linux.*

kejxu, 2019-09-03:
Just a quick update on the thread, we released [`0.6.0`](https://discourse.ros.org/t/updating-the-vs-code-extension-for-ros-to-0-6-0/10508/2) last week (initial release for attach-debug and ROS2 support). Would love to know how that works =)

Meantime, launch-debug (debugging roslaunch) is currently work-in-progress.

mElBalkini, 2019-09-06:
Attach debugger works perfectly. I tested on Linux.
I also used [this trick](https://stackoverflow.com/questions/663449/what-is-the-best-way-to-attach-a-debugger-to-a-process-in-vc-at-just-the-right) to wait till the debugger is attached.
Waiting for the automated launch-debug feature.
Thanks for the quick response and implementation

rcabg, 2019-09-24:
@kejxu sorry for the delay. Attach to process works really well, tested on Ubuntu 16.04. Congratulations for the work! I will stay tuned for the launch-debug.

rickstaa, 2019-10-02:
Additionally, to the ways described by @mElBalkini, here are some additional methods to make the debugger wait that worked better in my specific situation.

- [Wait for gdb to atach](https://stackoverflow.com/questions/11255064/wait-for-gdb-to-attach)
- [Set breakpoint in c or c++ code programmatically for gdb on linux](https://stackoverflow.com/questions/4326414/set-breakpoint-in-c-or-c-code-programmatically-for-gdb-on-linux)

fjp, 2020-03-19:
How is it possible to debug tests?

I am using this [tasks.json](https://github.com/ms-iot/vscode-ros/issues/114#issuecomment-526222538) from #114 to build test in debug type. This works so far. 

I also setup the "ROS: Attach" debug configuration in the launch.json.

When I run the "catking this test debug" task and try to attach to the "catkin" or "make" process (not sure which one is correct) using F5 I get:
```
 $  env sh /tmp/Microsoft-MIEngine-Cmd-p6vzc5rj.2en 
Superuser access is required to attach to a process. Attaching as superuser can potentially harm your computer. Do you want to continue? [y/N] env sh /tmp/Microsoft-MIEngine-Cmd-oyd3c0gb.9r6
```
When I accept this and enter my super user password the breakpoints get greyed out:

[image]

Note that I've also added the following to my code to find the time to attach to the process.

```cpp
int i = 0;

while (i == 0)
{
    usleep(100000);  // sleep for 0.1 seconds
}
```

Did I attach to the wrong process? Which one should I use while the test is running?

[image]

[image]

fjp, 2020-03-20:
I got to the point where I can stop at the breakpoint. The main problem was that I didn't attach to the correct process because the ROS package I am working with (`ros_controllers/diff_drive_controller` from ROS control) is not my own and I therefore don't know the executables exactly.

However here is the `catkin config` I used, notice the "Additional CMake Args (I guess it was important to set this to Debug):

```bash
$ catkin config
---------------------------------------------------------
Profile:                     debug
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/fjp/git/ros_ws
---------------------------------------------------------
Build Space:        [exists] /home/fjp/git/ros_ws/build
Devel Space:        [exists] /home/fjp/git/ros_ws/devel
Install Space:      [unused] /home/fjp/git/ros_ws/install
Log Space:          [exists] /home/fjp/git/ros_ws/logs
Source Space:       [exists] /home/fjp/git/ros_ws/src
DESTDIR:            [unused] None
---------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Debug
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------
Whitelisted Packages:        diff_drive_controller
Blacklisted Packages:        None
---------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------
```

I built the tests for the `diff_drive_controller` package (inside the package src folder) with:

```bash
catkin run_tests --no-deps --this
```

After that I executed a single test with:

```bash
rostest diff_drive_controller diff_drive_controller.test
```

Then I observed new processes with `htop` and saw a `diffbot` process. Quickly attaching to this process after starting the test let's the debugger stop at the breakpoint :-)

Thank you so far implementing the "Attach to process" feature :-)

Are there any plans to enable debugging tests (gtest and rostest) when they execute?

And I still need to find a way to just build the tests with `catkin` without executing them directly. Does someone know how this is done?

fjp, 2020-03-20:
I'd be also happy to see some documentation on how to setup ros debugging for nodes and tests. Please let me know if this exists already and excuse me if I missed it.

brunoeducsantos, 2020-10-27:
@fjp how have you managed to step in the break points?

ooeygui, 2020-10-27:
I'm going to close this issue as launch debugging does work. We need to add launch debugging of rostest https://github.com/ms-iot/vscode-ros/issues/330

closed 2020-10-27 by ooeygui