ms-iot/vscode-ros #97: [bug] Xacro file not rendered correctly

PymZoR, 2019-08-09:
**describe the bug (*actual behavior*)**
I have a standard xacro file to create an URDF (that unfortunately cannot be shared here).
It contains meshes on the `.stl` format for visuals, and some `gazebo` tags.

Running `ROS: Preview URDF` show a partial model.
Running xacro manually, and previewing the resulting URDF displays correctly.

**expected behavior**
The full model should be rendered.

**screenshots**
Rendering the xacro file :
[image]

Rendering the resulting URDF after manual conversion :
[image]

seanyen, 2019-08-14:
@PymZoR Hi can you elaborate more how did you do the manual `xacro` conversion? I am interested in if `vscode-ros` is doing it differently.

PymZoR, 2019-08-17:
@seanyen  `rosrun xacro xacro [path-to-xacro-file] > output.urdf`

ooeygui, 2019-09-24:
@PymZoR Could you share the location of the xacro file and support files? I'd like to debug this locally.

PymZoR, 2019-09-26:
I found the cause of the bug. Our xacro file was in fact called `model.urdf`.

An elegant fix would be to look after the tag `<robot xmlns:xacro="http://www.ros.org/wiki/xacro">` instead of the file extension.

ooeygui, 2019-09-26:
@PymZoR Thanks for clarifying. I changed this to an improvement, as it would require deep introspection.

ooeygui, 2022-10-13:
Related to https://github.com/ms-iot/vscode-ros/issues/746

ooeygui, 2023-05-04:
This has been corrected in v0.9.4

closed 2023-05-04 by ooeygui