ms-iot/vscode-ros #114: [bug] Catkin build tasks not working

beetleskin, 2019-08-26:
**general information**
Ubuntu 18.04; using catkin tools, vscode 1.37.1.

**what is the bug**
When running `Configure Default Build Task` and then selecting `catkin: make`, a new entry is added to _tasks.json_:
```
{
            "type": "catkin",
            "problemMatcher": [
                "$catkin-gcc"
            ],
            "group": {
                "kind": "build",
                "isDefault": true
            }
        }
```

For the type, vscode complains about "Value is not accepted". For "problemMatcher: "Incorrect type. Expected object".

**expected behavior**
The creation of a valid and working _catkin build_ task

beetleskin, 2019-08-29:
Alright. With the original plugin, I've been using those:
```javascript
{
    "version": "2.0.0",
    "tasks": [
        {
            "label": "catkin all default",
            "type": "shell",
            "command": "catkin build --profile default --continue",
            "problemMatcher": {
                "base": "$gcc",
                "fileLocation": [
                    "absolute"
                ]
            },
            "group": "build"
        },
        {
            "label": "catkin all debug",
            "type": "shell",
            "command": "catkin build --profile debug --continue",
            "problemMatcher": {
                "base": "$gcc",
                "fileLocation": [
                    "absolute"
                ]
            },
            "group": "build"
        },
        {
            "label": "catkin this default",
            "type": "shell",
            "command": "cd ${fileDirname} && catkin build --profile default --this --no-deps",
            "problemMatcher": {
                "base": "$gcc",
                "fileLocation": [
                    "absolute"
                ]
            },
            "group": {
                "kind": "build",
                "isDefault": true
            }
        },
        {
            "label": "catkin this debug",
            "type": "shell",
            "command": "cd ${fileDirname} && catkin build --profile debug --this --no-deps",
            "problemMatcher": {
                "base": "$gcc",
                "fileLocation": [
                    "absolute"
                ]
            },
            "group": "build"
        },
        {
            "label": "catkin this test default",
            "type": "shell",
            "command": "cd ${fileDirname} && catkin run_tests --profile default --this --no-deps",
            "problemMatcher": {
                "base": "$gcc",
                "fileLocation": [
                    "absolute"
                ]
            },
            "group": "build"
        },
        {
            "label": "catkin this test debug",
            "type": "shell",
            "command": "cd ${fileDirname} && catkin run_tests --profile debug --this --no-deps",
            "problemMatcher": {
                "base": "$gcc",
                "fileLocation": [
                    "absolute"
                ]
            }
        },
        {
            "label": "catkin clean",
            
            "type": "shell",
            "command": "catkin clean --all-profiles"
        }
    ]
}
```

Do you mean those type of tasks? Or how do I compile with the plugin?

orzechow, 2021-01-08:
With VSCode 1.52.1 and vscode-ros v0.6.5, this is still an issue.

When creating a build task for a catkin tools workspace using the plugin (`CTRL+SHIFT+P` → `Tasks: Configure Default Build Task`), this `tasks.json` is created:

```json
{
	"version": "2.0.0",
	"tasks": [
		{
			"type": "catkin",
			"args": [
				"--workspace",
				"/path/to/my/catkin_ws"
			],
			"problemMatcher": [
				"$catkin-gcc"
			],
			"group": {
				"kind": "build",
				"isDefault": true
			},
			"label": "catkin: build"
		}
	]
}
```

As a result triggering a build fails:

```
> Executing task: catkin --workspace /path/to/my/catkin_ws <

usage: catkin [-h] [-a] [--test-colors] [--version]
              [--force-color | --no-color]
              [build | clean | config | create | env | init | list | locate | profile]
              ...

Error: Unknown verb '/path/to/my/catkin_ws' provided.
The terminal process "/bin/bash '-c', 'catkin --workspace /path/to/my/catkin_ws'" terminated with exit code: 1.
```

Clearly, it should have called `catkin build`. These changes in `tasks.json` fix the problem (and nicely set the working directory, to support stuff like `catkin build --this --no-deps` as mentioned by @beetleskin ):

```json
{
	"version": "2.0.0",
	"tasks": [
		{
			"label": "catkin: build all",
			"type": "catkin",
			"args": [
				"build",
				"--continue"
			],
			"problemMatcher": [
				"$catkin-gcc"
			],
			"group": "build",
			"options": {"cwd": "${fileDirname}"}
		}
	]
}
```

closed 2022-10-13 by ooeygui