ms-iot/vscode-ros #116: vscode-ros not sourcing workspace

rickstaa, 2019-08-27:
**general information**
\<OS version, Ubuntu? Windows?\> Ubuntu 18.04
\<Visual studio code version\> 1.38.0-insider
\<Version of the plugin\> 0.5.0
\<python environment\> miniconda3 (py2.7)

**what is the bug**
When I try to use the rosrun debug function (F5 after roscore is started) nothing happens. Further when trying to debug the workspace `devel/setup.bash` is not sourced before debugging the python node. As a result I can not import modules from other rospackages.

**expected behavior**
Having the ability to debug the rosnode on ubuntu 18.04. Further I also expected the Plugin to source the `devel/setup.bash` file before starting the debugger.
=

rickstaa, 2019-09-01:
@kejxu Thanks for looking into my problem and for rebuilding the debugger. I'm happy to perform some tests if the debugger is ready.

kejxu, 2019-09-03:
You're very welcome @rickstaa, we're always trying to understand more about different kinds of problems! We recently just announced [attach-debug](https://discourse.ros.org/t/updating-the-vs-code-extension-for-ros-to-0-6-0/10508), this would hopefully make debugging a ROS (C++/Python) node by attaching to the process easier =) (if it does not, we're more than happy to look into the problem)

Meantime, debugging a ROS node at entry is still a work-in-progress.

rickstaa, 2019-09-03:
@kejxu Wow thanks that work seamlessly for attaching the process. If you could add ROS conda/virtualenv setup.bash file sourcing my ROS Development environment would be complete. I'm unsure if it is a problem with your extension or the general way visual studio behaves so I also posted an issue on the vscode github [gitub.com/microsoft/vscode](https://github.com/microsoft/vscode/issues/79932).

kejxu, 2019-09-03:
> Wow thanks that work seamlessly for attaching the process.

Yay! That just sounds wonderful =)

> If you could add ROS conda/virtualenv `setup.bash` file sourcing my ROS Development environment would be complete. I'm unsure if it is a problem with your extension or the general way visual studio [...]

I am not 100% sure what your ROS dev environment looks like, but I agree it would be worthwhile looking into how ROS should work with `venv` (there're likely work we need to do for this extension, but we need to dive further into the usage before that). The issue you posted in `vscode` looks like a very good starting point! Thanks for opening that ticket!

sofu456, 2019-09-09:
maybe you config source */setup.bash in .bashrc,so you will reset you path every time you restart

sofu456, 2019-09-09:
maybe you config source */setup.bash in .bashrc,so you will reset you path every time you restart. and so this is not a bug.

ps: i get the source code,find the termial will be change every time it start

sofu456, 2019-09-09:
maybe you config source */setup.bash in .bashrc,so you will reset you path every time you restart. and so this is not a bug.  just delete this config,and use the vscode purge setup.

ps: i get the source code,find the termial will be change every time it start

rickstaa, 2019-09-26:
@sofu456 Thanks for your answer. That is indeed what I have been doing up till now. I, however, think it could be very valuable to incorporate auto sourcing into the extension.

```
# Source ros setup.bash if present
if [ -f '../devel/setup.bash' ]; then . "../devel/setup.bash";fi
``

rickstaa, 2019-09-26:
As stated by [@isidorn][1] in this [vscode GitHub issue](https://github.com/ms-iot/vscode-ros/issues/116) this feature is currently not yet supported. In the meantime, people can achieve the desired behaviour by adding the following code to their `.bashrc`

```
# Source ros setup.bash if present
if [ -f '../devel/setup.bash' ]; then . "../devel/setup.bash";fi
```
  [1]: https://github.com/isidorn

sofu456, 2019-09-27:
> @sofu456 Thanks for your answer. That is indeed what I have been doing up till now. I, however, think it could be very valuable to incorporate auto sourcing into the extension.
> 
> ```
> # Source ros setup.bash if present
> if [ -f '../devel/setup.bash' ]; then . "../devel/setup.bash";fi
> ```

right,i has commit a pullrequest,this may solve this question as well

[https://github.com/ms-iot/vscode-ros/pull/123](url)

rickstaa, 2019-09-27:
@sofu456 Thanks for the pull request! I looked through the code changes but I'm unsure how they automatically source the ROS workspace. Does it still need the above-mentioned workaround?

sofu456, 2019-09-27:
> @sofu456 Thanks for the pull request! I looked through the code changes but I'm unsure how they automatically source the ROS workspace. Does it still need the above-mentioned workaround?

you can also add the config into bashrc file,and the ros plugin will run source ../devel/setup.bash after the terminal is launch finish.

the ros plugin run this command before the teiminal launch finish before,so it cause this problem.
now it is ok,but i was puzzle the manager has ready update the plugin in vscode.

rickstaa, 2019-09-30:
@sofu456 Ah I see thanks again!

beetleskin, 2019-11-28:
I'm missing this feature (source workspace before running python) since the very first day of the plugin, so I'm super i favor of this :)

> > @sofu456 
> you can also add the config into bashrc file,and the ros plugin will run source ../devel/setup.bash after the terminal is launch finish.

That's often the answer from many IDEs (QtCreator, CLion, Eclipse, ..) but I think this is rather ugly: Often, we work on multiple workspaces and switching should be easy. Sourcing and running the editor from commandline is not easy, also not editing your bashrc a dozen times a day. If I run any IDE, it'd be nice if it knows my workspace and what comes with it.
Currently, the QtCreator (@Levi-Armstrong) plugin solves this nicely, but it also took some time :)

rickstaa, 2020-05-21:
@kejxu Since @beetleskin also showed an interest in this feature is it possible to change the bug label to a feature request label?

kejxu, 2020-05-21:
> @kejxu Since @beetleskin also showed an interest in this feature is it possible to change the bug label to a feature request label?

@seanyen

tlobig, 2020-08-27:
I would also need this feature. even adding the sourcing to .bashrc didn't work - standard "source /opt/noetic/ ..." does, but not sourcing from my install within the catkin workspace. It does work for the terminal window, but when I start code from there, the extension does still not recognize my own packages. So I can't ros launch for debugging and i can't use roslaunch from the command palette to start my own nodes.

ooeygui, 2022-10-13:
We have added support for custom startup scripts and better autodetection. Given the age of this bug, I'm going to close. Please open a new issue if none of the existing supported paths work.

closed 2022-10-13 by ooeygui

starball5, 2024-08-22:
> As stated by [@isidorn](https://github.com/isidorn) in this [vscode GitHub issue](https://github.com/ms-iot/vscode-ros/issues/116) the prelaunch task feature is currently not yet supported.

Pretty sure you meant to link to https://github.com/microsoft/vscode/issues/79932#issuecomment-535437634 instead.