ms-iot/vscode-ros #132: Ros python debugger gives timeout

rickstaa, 2019-09-24:
**general information**
Ubuntu: 18.04
Plugin v: 0.6.0

**what is the bug**
The python ROS debugger keeps giving me the following timeout error after 1 min of debugging:

```
Exception in thread ptvsd.Server:
Traceback (most recent call last):
  File "/home/ricks/miniconda3/envs/py2_auto_grasp/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/home/ricks/miniconda3/envs/py2_auto_grasp/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/ricks/miniconda3/envs/py2_auto_grasp/lib/python2.7/site-packages/ptvsd/wrapper.py", line 536, in process_messages
    raise exc
timeout: timed out
```

**expected behavior**
No timeout.

rickstaa, 2019-09-27:
@seanyen Thanks for your response. I checked the version of the `ptvsd` debugger and it is currently at version 4.3.2. As this is the latest version I could find for my python 2.7 conda environment the problem still persists.

seanyen, 2019-10-08:
> @seanyen I just noticed that it only happens when I'm debugging inside a ROS service.

Interesting. Do you mind elaborating more on this? Any quick example that we can try?

rickstaa, 2019-10-11:
@seanyen, Of course, I will make an example somewhere next week.

rickstaa, 2019-10-11:
@seanyen In the meantime, and as a workaround, I noticed that the timeout doesn't happen when I use the following code to wait for the debugger.

```
## DEBUG: WAIT FOR PTVSD DEBUGGER ##
import ptvsd
ptvsd.wait_for_attach()
## ------------------------------ ##
```

rickstaa, 2021-05-15:
@seanyen For future reference now that VsCode has switched to using [debugpy](https://visualstudiomagazine.com/articles/2020/03/20/vs-code-python-debugger.aspx) the work-around should be updated to:

```python
# DEBUG: WAIT FOR USER TO ATTACH DEBUGGER
i = 1
  while i == 1:
      time.sleep(0.5)  # Change i to 2 to get out of loop
# ------------------------------
```

I also tried the using the `wait_for_client` method of the [debugpy](https://github.com/microsoft/debugpy) debugger but it required more worth since I then had to set a fixed port for the debugger to listen to.

```
# DEBUG: WAIT FOR DEBUGPY DEBUGGER
import debugpy
debugpy.wait_for_attach()
# ------------------------------ 
```

rickstaa, 2021-06-02:
@seanyen Closing this since it seems to be fixed in v0.6.8. I'm not 100% sure, but I haven't seen the error for a while. @seanyen Closing this since it seems to be fixed in v0.6.8. If other people run into this error, feel free to comment, and I will reopen. Or create a new issue request.

closed 2021-06-02 by rickstaa