ms-iot/vscode-ros #139: Enhancement: Add documentation about debugging ROS Nodes

michaelajax, 2019-10-17:
It would be good if we created documentation about building nodes in a way that allows a debugger to be attached, as well as documentation about how to attach a debugger.

Some things to consider:
- Documentation about adding a wait for debugger to the node source. quick summary is:
    - add `#include "windows.h"` to the node source code
    - add the following lines right at the start of the `main()` routine:
        ```
        while (!IsDebuggerPresent())
		Sleep(100);
        ```
- Documentation about starting the node, and then how to Attach VS to it:
    - rebuilding node with catkin_make
    - info on setting breakpoints in VS
    - info and screenshots of attaching VS to a running process (such as ensuring native debugging is selected)

seanyen, 2019-10-17:
@michaelajax Please excuse my late response. The debugging experience is what our ROS extension tries to tackle. I believe now we have some native ROS nodes debug support without the `sleep()` trick, but just doesn't have good documentation for that.

cc @kejxu maybe that's something we should add into our backlog.

mcevoyandy, 2020-04-24:
I've recently switched to using VSCode for my ROS dev and am liking it, but I would also like to see more documentation. It's frustrating that there isn't more. I still debug in my terminal window with GDB...

ooeygui, 2022-10-13:
We have added more documentation and videos on debugging in multiple environments. Check them out :-)

closed 2022-10-13 by ooeygui