ms-iot/vscode-ros #148: Conout load the URDF viewer [bug]

Geekgineer, 2019-11-06:
**General information**

- Ubuntu 16.4
- The version of the plugin 0.6.1
- code --version
- 1.39.2
- 6ab598523be7a800d7f3eb4d92d7ab9a66069390
- x64

**Bug**
cannot load the URDF viewer
[image]

**expected behavior**
wait some time for the model to be loaded

**additional context**
The packages load correctly in RViz and Gazebo
also, I am sure that I have sourced the ws and ROS by default in .bashrc

source /opt/ros/kinetic/setup.bash
source ~/catkin_ws/devel/setup.bash

seanyen, 2019-11-06:
Thanks for the reporting. One quick question, do you see your `ROS_PACKAGE_PATH` configured correctly if you are checking it from the terminal inside the `vscode`? I just wanted to make sure `vscode` doesn't see anything different.

kejxu, 2019-11-06:
thanks a lot for reporting this issue! another quick check, have you by any chance tried executing the xacro command by itself in the terminal?

asking because it seems like this is happening due to an error from `xacro`

Geekgineer, 2019-11-06:
> Thanks for the reporting. One quick question, do you see your `ROS_PACKAGE_PATH` configured correctly if you are checking it from the terminal inside the `vscode`? I just wanted to make sure `vscode` doesn't see anything different.

Thank you for the quick support! really appreciate that, 
yes, these work perfectly and show me all the directors related to ROS.
```sh
$ echo $ROS_PACKAGE_PATH 
$ echo $ROS_ROOT 
```
[image]

Geekgineer, 2019-11-06:
> thanks a lot for reporting this issue! another quick check, have you by any chance tried executing the xacro command by itself in the terminal?
> 
> asking because it seems like this is happening due to an error from `xacro`

Thank you for your efforts! yes, it works properly.
```sh
  $ xacro legonplate.urdf.xacro 
```

[image]

ooeygui, 2023-05-04:
This has been corrected

closed 2023-05-04 by ooeygui