ms-iot/vscode-ros #149: [feature] Can I use "-enable-pretty-printing" option in ROS Attach?

clobotcalc100, 2019-11-07:
- Ubuntu : Desktop 16.04 LTS
- ROS : Kinetic Kame
- VSCode : 1.39.2 (+ROS extensions)

**Can I use "-enable-pretty-printing" option in configurations ( launch.json )?**
I want to see **plan** (std::vector<geometry_msgs::PoseStamped> variables.

[image]
[image]

Thanks.

kejxu, 2019-11-11:
thanks for reporting this!

quick question, is it possible to add this option in a [C++ debug configuration](https://code.visualstudio.com/docs/cpp/launch-json-reference)? Like using the `miDebuggerArgs` property?

if that is possible, this might be duplicate with https://github.com/ms-iot/vscode-ros/issues/122

sweetim, 2020-04-09:
this would be useful, at the moment either

- write own launc file as suggested (https://answers.ros.org/question/313371/vscode-debug-cpp-ros-node/)
- or manually input `-enable-pretty-printing` in `debug console` (https://qiita.com/taka_horibe/items/50e4659e88d1ee3cd04a)

as the current configuration do not allow override of the `miDebuggerArgs` when using `type: "ros"`

ooeygui, 2022-10-13:
This has been added to the extension

closed 2022-10-13 by ooeygui