ms-iot/vscode-ros #151: [bug]

15800436469try, 2019-11-11:
when I start debug,it shows a wrong information  "Cannot convert undefined or null to object" ,but my steps is right

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "ROS: Launch",
            "request": "launch",
            "target": "/home/luochao/catkin_ws/src/beginner_tutorials/launch/turtleminic.launch",
            "type": "ros"
        }
    ]
}

rcabg, 2019-11-28:
Same problem here. So far unable to do a roslaunch debug since always appears that message.

Ubuntu 16.04
vscode: 1.40.1
ROS extension: 0.6.2
ROS Kinetic

Hyeokreal, 2020-07-14:
Same problem here.

matthias-mayr, 2020-07-16:
I had the same issue with codium 1.46.1-1592564058 on Ubuntu 18.04. I upgraded `codium` to the latest version 1.47.1-1594810499. I was not able to find this extension in the marketplace after the upgrade anymore, so I downloaded the `vsix` file from the website and installed it manually. 
But now this error message is gone.

ooeygui, 2021-11-06:
Closing, I cannot reproduce this. If you are able to get a stack trace, or a pointer to a repo or which can reproduce this, please reopen

closed 2021-11-06 by ooeygui