ms-iot/vscode-ros #152: c++ ros nodes not showing up in ROSRUN menu [bug]

rickstaa, 2019-11-13:
**general information**
OS: Ubuntu 18.04
Version: 1.41.0-insider
Commit: 0d728c31ebdf03869d2687d9be0b017667c9ff37
Date: 2019-11-12T05:32:12.922Z
Electron: 6.1.4
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Linux x64 5.0.0-32-generic

**what is the bug**
When I try to use the `>ROS: Run a ROS executable (rosrun)` command only my python nodes are showing up. I can however manually start the C++ ROS node in an ROS terminal and use the attach a debugger.

**expected behavior**
That my C++ ROS nodes are listed in the rosrun menu.

kejxu, 2019-11-13:
Thanks for reporting this!

under the hood the extension tries to find executables (on Linux) using the `find -type f -executable` [command](https://github.com/ms-iot/vscode-ros/blob/master/src/ros/ros1/ros1.ts#L107). Just a quick check, does the `octomap_server_node` binary have the `-x` flag added?

rickstaa, 2019-11-14:
@kejxu Thanks a lot for your response! I am not sure if I fully understand your question. I did add the `-DCMAKE_BUILD_TYPE=RelWithDebInfo` flag to the catkin config. Do I also need to add an extra flag inside the CMAKELIST.txt file?

kejxu, 2019-11-14:
not exactly, I was wondering about the permissions (`chmod`) of the binary. try running this command:

```bash
ls -l <path/to/octomap_server_node>
```

any `x` in the output ([sample](https://www.computerhope.com/unix/uchmod.htm)) would mean the binary could be executed.

rickstaa, 2019-11-15:
@kejxu Thanks for your answer! I did not give my user execution permissions as I didn't think this was needed as I could use the  `rosrun octomap_server octomap_server_node` command from within a terminal. Changing the x permissions indeed fixed my problem.

rickstaa, 2019-11-15:
@kejxu  I did some further testing, and I now get the following error while trying to use the extensions `rosrun` command.

```
/home/rstaa/panda_autograsp_ws/src/panda_autograsp/src/talker.cpp: line 1: /bin: Is a directory
/home/rstaa/panda_autograsp_ws/src/panda_autograsp/src/talker.cpp: line 2: syntax error near unexpected token `('
/home/rstaa/panda_autograsp_ws/src/panda_autograsp/src/talker.cpp: line 2: ` * Copyright (C) 2008, Morgan Quigley and Willow Garage, Inc.'
```

I think this is caused by the fact that the node is called as `rosrun beginner_tutorials talker.cpp` instead of `rosrun beginner_tutorials talker`.

kejxu, 2019-11-15:
thanks for updating on this!

passing `talker.cpp` into the rosrun command should not happen (as long as we're choosing the executable itself, `talker`, not `talker.cpp`).

Since the `rosrun` command from the extension is actually executing a `rosrun` command in the terminal, the actual command and arguments should be printed on the terminal too, could you also take a look at the terminal output?

rickstaa, 2019-11-18:
@kejxu I just did some checks, and it appears that the `.cpp` Extension is forwarded to the console command when the `x` flag is enabled. The console output is:

```
rosrun beginner_tutorials talker.cpp 
``

the error message I get in that situation is:

```
home/rstaa/catkin_ws/src/vscode-ros_issue152/src/talker.cpp: line 6: /bin: Is a directory
/home/rstaa/catkin_ws/src/vscode-ros_issue152/src/talker.cpp: line 7: CMakeLists.txt: command not found
/home/rstaa/catkin_ws/src/vscode-ros_issue152/src/talker.cpp: line 8: include/: Is a directory
/home/rstaa/catkin_ws/src/vscode-ros_issue152/src/talker.cpp: line 9: syntax error near unexpected token `('
/home/rstaa/catkin_ws/src/vscode-ros_issue152/src/talker.cpp: line 9: `int main(int argc, char **argv)'
```

Below is a gif showing the behavoir.

![debug_problem](https://user-images.githubusercontent.com/17570430/69081149-ad992880-0a3d-11ea-8a92-f2a5301551ed.gif)

The example repository can be found [here](https://github.com/rickstaa/vscode-ros_issue152).

rickstaa, 2021-06-18:
@kejxu I will close this since this bug is also being investigated in the newer issue of @yassiezar.

closed 2021-06-18 by rickstaa