ms-iot/vscode-ros #154: command 'ros.XXXXX' not found in Win10

Konye666, 2019-11-18:
**general information**
\<Win10 64bit\>
\<0.6.2\>

Hello, I am a beginner in VS Code, Can the extension work in Windows Systems? Commands like 'ros.startCore' and 'ROS: Create Catkin Package'are all not found when using Ctrl+Shift+P to run. Seems that I need to open a catkin workspace first, but how can I open a catkin workspace in VS Code?

Thanks.

seanyen, 2019-11-19:
@Konye666 Hi! Thanks for the reporting. Assume that you have a catkin workspace under `c:\catkin_ws`, and you have to run `code` under the root of the workspace. And one thing to be noted is that you need to do at least once `catkin_make` to bootstrap your workspace, so ROS extension knows you are really under a catkin workspace.

Hence, for example, you probably will need to do the following:
```
:: change the directory to the root
pushd c:\catkin_ws

:: build the workspace at least once
catkin_make

:: now launch the vscode
code .
```

ooeygui, 2022-10-13:
it appears that vscode was not loaded on the root of a catkin workspace. We do have other issues to improve the "not found" case, so closing this one.

closed 2022-10-13 by ooeygui