ms-iot/vscode-ros #155: Unable to find ros packages

Nabel0721, 2019-11-19:
When opening my workspace, the extension is not able to find any of the ROS packages in that workspace. I have checked that running rospack find does list them all, and when using the terminal from the ROS: Create Terminal command the workspace's devel/setup.bash is already sourced. Is there something I'm missing that needs to be done before it sees the workspace?

seanyen, 2019-11-19:
@Nabel0721 Hi! Thanks for reporting. Did you start the `code` from the root of your workspace? And what do you mean by not able to find any ROS packages? Are you able to share a screenshot?

Nabel0721, 2019-11-19:
I launched code from the app picker on the start menu, and then opened the root of my workspace with "Open Folder". I can share a screenshot tomorrow, but when I run the command to run a launch file or executable, the bar to enter a package appears but there are no suggestions and it doesn't accept a package name if I type it in directly.

namals, 2019-11-20:
I faced the same issue when trying to setup the launch debugging. It threw the error "Cannot convert undefined or null to object" mentioned in https://github.com/ms-iot/vscode-ros/issues/151

After looking at @seanyen's response about starting `code` from the root of the workspace, I noticed that I have configured the workspace for `code` to be in ~/catkin_ws/src rather than ~/catkin_ws. ( ~/catkin_ws is the root of the catkin workspace ) 

Creating a new `code` workspace in the root of the catkin workspace (i.e. ~/catkin_ws) resolved this issue. 

Note: this creates the .vscode directory in ~/catkin_ws. But now it complains about too many files to keep track for changes and sends me to https://code.visualstudio.com/docs/setup/linux#_visual-studio-code-is-unable-to-watch-for-file-changes-in-this-large-workspace-error-enospc

Nabel0721, 2019-11-20:
Here's a small video of what's happening. I am at the root of my workspace (which is named beta), and I have already built with catkin_make. 

[ROS Extension.mp4.zip](https://github.com/ms-iot/vscode-ros/files/3869151/ROS.Extension.mp4.zip)

Github won't let me post an mp4, so it's in the zip archive. Hopefully this'll help

Nabel0721, 2019-11-21:
So, after thinking about it for a while, my gut tells me that the extension is not getting the same environment as my terminal is. It not only cannot find my packages, but also the builtin ros packages. Is there something I can do to point the extension to the place where the packages are located, like how the terminal knows because of running setup.bash?

Nabel0721, 2019-11-21:
From what I can tell the only place this issue could be coming from is in the `ros1.ts` file under either the `getPackageNames()` function or the `getPackages()` function. Since the extension shows the input for a package but never loads in the packages, I think it's waiting on a promise that is never resolved. I'm not sure how this is happening, since the command it's running, `rospack list`, is very quick, but that's where the issue should be.

Nabel0721, 2019-11-21:
After looking through the roscore commands in [here](https://github.com/ms-iot/vscode-ros/blob/master/src/ros/ros1/core-helper.ts#L19) (since those still work for me), I found that the environment is obtained through the extension module, rather than being passed through directly [right here](https://github.com/ms-iot/vscode-ros/blob/master/src/ros/ros1/ros1.ts#L23). In the functions I mentioned in the last message, the environment is passed directly into the functions. Is there any reason for these to be different, and if so could it be the cause of the command hanging?

tlobig, 2020-08-27:
I'm stuck with the same problem. Creating the code workspace within the catkin workspace did nothing for me, the launch option does not work since my internal package, i.e. within the given catkin workspace, is not found. I would love to configure the catkin workspace for the extension, then it would be transparent where it is looking for packages.

chisarie, 2020-11-06:
Hi, I had the same problem. The solution I found is that you need to launch vscode from a sourced terminal window:
```bash
cd catkin_ws
source devel/setup.bash
code
```

FehlerNotFound, 2020-12-01:
I have the same issue and finally solved it!
you should:
cd ..._ws
source devel/setup.bash
cd src/
roslaunch (movelt assistant file)

ooeygui, 2022-10-13:
We have some documentation and videos on how to launch the VSCode extension so it can property configure the workspace. Please take a look at https://youtu.be/uqqHgYsskJI

closed 2022-10-13 by ooeygui