ms-iot/vscode-ros #156: [feature] automatic ROS master activation when doing roslaunch debugging.

seanyen, 2019-11-20:
Currently when using `roslaunch` debug, the extension assumes the ROS master is up and running. However we can make this little bit smarter than that, and decide to run ROS master on behave of the users if there is no any running instance.

mickare, 2020-01-23:
Hey @seanyen, i'm using a workaround to start the ROS master... maybe it helps you.

I am using the [Remote Development extension pack](https://marketplace.visualstudio.com/items?itemName=ms-vscode-remote.vscode-remote-extensionpack) and I configured it to automatically start the roscore when the develop-environment-container is started.

*.devcontainer/devcontainer.js*
```
{"name": "ROS
	"context": "..",
	"dockerFile": "Dockerfile",
	"remoteUser": "myStaticUser",
	"runArgs": [
		"--cap-add=SYS_PTRACE",
		"--security-opt",
		"seccomp=unconfined",
		"--tty"
	],
	"overrideCommand": false,
	"workspaceMount": "source=${localWorkspaceFolder},target=/catkin_ws,type=bind,consistency=delegated",
	"workspaceFolder": "/catkin_ws",
"mounts": [
		// Avoiding extension reinstalls on container rebuild
		"source=proseco-extension,target=/root/.vscode-server/extensions,type=volume",
		// And/or for VS Code Insiders
		"source=proseco-extension-insiders,target=/.vscode-server-insiders/extensions,type=volume",
		// Mount ssh keys
		"source=${env:HOME}${env:USERPROFILE}/.ssh,target=/root/.ssh,type=bind",
		"source=${env:HOME}${env:USERPROFILE}/.ssh,target=/home/proseco/.ssh,type=bind",
		// Display
		"source=/tmp/.X11-unix,target=/tmp/.X11-unix,type=bind"
	],
	"settings": {
		"terminal.integrated.shell.linux": "/bin/bash",
		"terminal.integrated.shellArgs.linux": [
			"-l"
		],
		"ros.distro": "melodic"
	},

*.devcontainer/ros_entrypoint.sh*
```
#!/bin/bash
set -e

# setup ros environment
source "/opt/ros/$ROS_DISTRO/setup.bash"

# exec "$@"

# Run commands with the container-user
exec gosu "myStaticUser" bash -l -c "$@"
```

* Dockerfile*
```

ooeygui, 2021-06-25:
This has been completed.

closed 2021-06-25 by ooeygui