ms-iot/vscode-ros #161: [feature] allow configuration of colcon build command

fmorgner, 2019-12-11:
**what is needed**
Currently, the generated build command for `colcon` builds simply executes `colcon build`. It would be nice if there was a way to specify additional parameters for the invocation of `colcon`, for example to specify additional/different event handlers.

**why is it needed**
The current implementation does not provide any flexibility as to how `colcon` is invoked. As an example, to get access to the build logs, one has to navigate the logs directory hierarchy. Allowing to specify additional options/parameters for the `colcon` invocation would allow to specify `--event-handlers console_cohesion+`, so that build output gets logged into the `Output` console.

seanyen, 2020-02-25:
Have you tried this? https://colcon.readthedocs.io/en/released/user/configuration.html#defaults-yaml 

Developers can create a `defaults.yaml` to manage their custom `colcon` build commands. You can create one under your workspace and set `COLCON_HOME` environment variables before running the code and it should work?

ooeygui, 2022-10-13:
We have some specializations of this as well, including debug type and mixins. Leaving as is