ms-iot/vscode-ros #169: [bug] Attach to python node not working

rickstaa, 2020-02-28:
**general information**
**OS version:** Ubuntu 18.04
**Plugin version:** 0.6.2

**what is the bug**
When trying to attach to a python ROS node nothing happens.

**expected behaviour**
The vscode ROS extension successfully attaches to the running python ROS node.

**steps to reproduce**
1. Clone the rospy example repository.
2. Build the catkin workspace
3. Source the catkin workspace
4. Start the ros talker node using `rosrun my_robot_pkg talker.py`
5. Start the ros listener node using `rosrun my_robot_pkg listener.py`
6. Add the ROS attach option to your launch.json file.
7. Try to attach to the ROS python node using python->listener.py.

seanyen, 2020-05-20:
@rickstaa Sorry for the long overdue. Is it still reproducible? I tried the following combination and I can attach and debug the Python node.

Platform: Ubuntu 18.04

```
# output from `code --list-extensions --show-versions`
ms-iot.vscode-ros@0.6.3
ms-python.python@2020.5.80290
ms-vscode.cpptools@0.28.0
```

rickstaa, 2020-05-21:
@seanyen Thanks a lot for the heads-up! The issue is indeed solved in:

```
Version: 1.46.0-insider
Commit: 6849775440496504e0677ffcf2a46d47ad9130f5
Date: 2020-05-15T05:29:39.210Z
Electron: 7.2.4
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Linux x64 5.4.26-050426-generic snap
ms-iot.vscode-ros@0.6.3
ms-python.python@2020.5.80290
ms-vscode.cpptools@0.28.1
```

closed 2020-05-21 by rickstaa

hangdonglin, 2022-05-29:
Hi, I have encountered the same situation, it cannot attach to the listener or talker node.
How did you solve it ? just by updating the vscode version?
Here is my vscode version infos:
Version: 1.67.2
Commit: c3511e6c69bb39013c4a4b7b9566ec1ca73fc4d5
Date: 2022-05-17T18:23:40.286Z
Electron: 17.4.1
Chromium: 98.0.4758.141
Node.js: 16.13.0
V8: 9.8.177.13-electron.0
OS: Linux x64 5.4.0-113-generic snap