ms-iot/vscode-ros #183: [bug] STL and DAE models will not load in VS code URDF preview.

rydb, 2020-04-13:
OS: 18.04 Ubuntu
Plugin Version: 0.6.3
ROS Version: Melodic

Bug: VS-code plugin will not render STL or DAE models in urdf preview

Current behavior: does not render models, but renders generic geometry
E.G:
Generic geometry like a cylinder will load
```
 <link name="base_link">
    <visual>
      <geometry>
        <cylinder length="0.01" radius="0.2"/>
      </geometry>
    </visual>
  </link>
```
STL and DAE models will not
```
<link name="body_link">
        <visual>
            <geometry>
                <mesh filename="file:///home/rydb/catkin_test/doggo.dae"/>
            </geometry>
        </visual>
    </link>
```

Steps to reproduce:
1. Create a new ROS folder with catkin:
   mkdir catkin && cd catkin && mkdir src && mkdir urdf && catkin_make

2. Copy and paste full URDF file(file code below) into a urdf file 

3. Change the mesh file line's filepath to a working DAE or STL file

4. load URDF preview with ctrl + p ">ROS: preview URDF"
 
Expected behavior: STL and or DAE models will load

What I've tried:

-. I've restarted VS code to see if that fixed it according to #93 , it did not.

Additional info:
The STL and DAE im testing with will load in Rviz but not in VS code, so i'm 99% sure it is a VS code problem, not a problem of the model. Also for some reason, the VS code generic geometry doesn't render color either. It just renders the generic geometry as black. 

RViz picture: [image]
VS Code picture: [image]

Full URDF file:

```
<?xml version="1.0"?>
<robot name="StlTest">
  <link name="base_link">
    <visual>
            <color rgba="0.1 0.1 0.1 1"/>
      <geometry>
        <cylinder length="0.01" radius="0.2"/>
      </geometry>
    </visual>
  </link>
   
    <link name="body_link">
        <visual>
        <origin xyz="0 0 0" rpy="0 0 0"/> 
            <geometry>
                <mesh filename="file:///home/rydb/catkin_test/doggo.dae"/>
            </geometry>
        </visual>
    </link>

    <joint name="base_to_right_leg" type="fixed">
      <parent link="base_link"/>
      <child link="body_link"/>
  </joint>
</robot>
```

Full roslaunch file for working Rviz demo:

```
<launch>

  <arg name="model" default="$(find urdf_tutorial)/urdf/01-myfirst.urdf"/>
  <arg name="gui" default="true" />
  <arg name="rvizconfig" default="$(find urdf_tutorial)/rviz/urdf.rviz" />

  <param name="robot_description" command="$(find xacro)/xacro $(arg model)" />

  <node if="$(arg gui)" name="robot_state_publisher" pkg="robot_state_publisher_gui" type="robot_state_publisher_gui" />
  <node unless="$(arg gui)" name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
  <node name="rviz" pkg="rviz" type="rviz" args="-d $(arg rvizconfig)" required="true" />

</launch>
```

ooeygui, 2020-04-13:
The VSCode plugin it isn't expecting a file:// path. It is expecting a package:// format path because the file should be relative to a package. Can you try modifying your URDF file?

rydb, 2020-04-14:
> The VSCode plugin it isn't expecting a file:// path. It is expecting a package:// format path because the file should be relative to a package. Can you try modifying your URDF file?

Tried to figure out how to do that but im now in a catch 22. the URDF tutorial im following on the official ROS website: http://mirror.umd.edu/roswiki/urdf(2f)Tutorials(2f)Building(20)a(20)Visual(20)Robot(20)Model(20)with(20)URDF(20)from(20)Scratch.html

has an example of a ROS urdf file with a package, but in order for the plugin to detect the folder, I have to run catkin_make, which requires it to not have a package.

Is there a ROS project I can quickly put into Visual studio code so I can determine if the problem is because I'm using file:// instead of package:// ?

closed 2020-04-16 by rydb