ms-iot/vscode-ros #184: [bug] C++ node not appearing in rosrun command when built with `catkin build`

yassiezar, 2020-04-14:
**general information**

OS: Mint 19.3, based on Ubuntu 18.04
Plugin: v0.6.3

**what is the bug**

When trying to run a C++ node with the >rosrun command, the available nodes in the package don't appear. This is similar to the issue [here](https://github.com/ms-iot/vscode-ros/issues/152). However, I've checked that that the executable is marked with an -x. 

This can be reproduced by building the workspace with `catkin build` and trying to launch the built node from VSCode. After some digging, I've found that the catkin_tools package installs a symlink to the node executable in the catkin_ws/devel/lib/ directory, while putting the actual executable in the catkin_ws/devel/.private/lib/ directory. This seems like the default behaviour and I can't seem to find an easy way around this. This additional hidden directory is what seems to be causing the executable search to fail.

**suggested fix**

Looking at plugin's executable search function [here](https://github.com/ms-iot/vscode-ros/blob/master/src/ros/ros1/ros1.ts#L107), I see that the `find` command will not follow symlinks, which causes it to not find the executable in my case where the package is build with `catkin build`. I would suggest adding the flag -L to the `find` command, which will allow the search to follow symlinks. It will not add the actual executable to the output list, but it will add the symlink at least

ooeygui, 2020-04-14:
Internally, vscode does this:
            const dirs = `catkin_find --without-underlays --libexec --share '${packageName}'`;
            command = `find $(${dirs}) -type f -executable`;

Can you execute these two commands on your environment and share the output?

yassiezar, 2020-04-15:
When searching for the executable "pcl_polygon_mesh_to_triangle_list", I've run the command `find (catkin_find --without-underlays --libexec --share 'pcl_polygon_mesh_to_triangle_list') -type f -executable`, which reports no results. Changing that to `find -L (catkin_find --without-underlays --libexec --share 'pcl_polygon_mesh_to_triangle_list') -type f -executable` (note the -L) reports exactly 1 result: the symlink to the executable. 

I've added the -L shown above to my local VSCode plugin and it is able to find and run the executable in VSCode as expected (using >rosun). I'd open a PR, but a 2-character PR seems a bit weird haha

ooeygui, 2020-04-15:
Thank you for the investigation!

yassiezar, 2022-05-30:
I think this was resolved by #321 so marking closed

closed 2022-05-30 by yassiezar

ooeygui, 2022-05-30:
Thank you for closing the loop!