ms-iot/vscode-ros #196: [bug] Unable to locate certain packages inside VSCode

nikhilprakash99, 2020-04-29:
**general information**
OS version : Ubuntu 18.04 with ROS Melodic
VSCode :  1.44.2 ( ff915844119ce9485abfe8aa9076ec76b5300ddd x64 )
Plugin : 0.6.3

**Bug Description**
Certain Packages ( in my case px4 ) are not getting located in the ROS environment inside VSCode. This package is to be sourced and set up automatically on startup through the zsh startup script, but for some reason it is not working. when i manually run the command to source inside the terminal, it gets executed

**additional context**
please refer to the following video for a detailed description

Video Description : <https://youtu.be/9oULflynFQ4>
 
with reference to the above video,
The pkg to be sourced/located is px4
line 158 in .zshrc shows the relevant command for sourcing the pkg
```bash
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$FIRMWARE
```
when i run rospack list to locate the pkg, it does not show up. 
when i manually run the command, after that it shows up.

* this is a problem only inside Visual code since when i open a new terminal session outside, everything works normally
* i just set up all the plugins, this is the first time i am using the ros plugin

please help me resolve this issue, any help is much appreciated. if additional information is required, let me know

nikhilprakash99, 2020-04-29:
The other commands in .zshrc are working and are getting executed

ooeygui, 2022-10-13:
The VSCode extension is manually setting up the ROS environment and the local build environment. Customizations can be done through the `ros.rosSetupScript` vscode extension setting on a per environment basis, so you can mix in your own environment settings through that, and call other workspaces if needed.
I hope that helps!

closed 2022-10-13 by ooeygui