ms-iot/vscode-ros #205: [feature] documentation of ROS2 support

fkromer, 2020-05-03:
Would be great of you could document what functionality is provided w.r.t. ROS2. From https://github.com/ms-iot/vscode-ros/projects/2#column-5390805 this is not obvious.

gianlucacolucci, 2020-06-21:
It would be great to have not just a list of supported features in ROS2 but also some documentation in general for ROS2. Thank you!

ooeygui, 2020-08-08:
Thank you for the request, we'll be working on improving ROS2 support and documentation.

fkromer, 2020-08-08:
@ooeygui Thanks a lot!

nightduck, 2020-10-17:
A video demo would be useful. The [one given by Sean in 2019](https://vimeopro.com/osrfoundation/roscon-2019/video/379127667) says it works with ROS2, but does not cover how to use colcon or what to do with ROS2's lack of a roscore.

ooeygui, 2022-10-13:
I have added videos on vscode and documentation linked here - https://github.com/ms-iot/vscode-ros

tschuldigung, 2022-10-14:
Very nice - thank you!