ms-iot/vscode-ros #206: [bug] Handle anonymous nodes when running a launch file

rhaschke, 2020-05-05:
Obviously the extension determines the list of ROS nodes started by a launch file and subsequently determines the cmdlines to start those, first using `roslaunch --nodes` and then `roslaunch --args`. However, this mechanism fails if a ROS node is using an anonymous name (e.g. `name="$(anon rviz)"`), because the full name randomly changes between invocations of roslaunch.

ooeygui, 2020-08-08:
Thank you for the report.

roslaunch itself doesn't provide a generic way to handle autolaunching, which is why we took the approach that we did. In order to support this request, we would need to amend roslaunch to expose the details of launch.

I'm going to leave this open, but this is not something we can support from the extension without changes to ros itself.

rhaschke, 2020-08-08:
Is it required that you start the individual nodes yourself?
Could you instead just call roslaunch as usual and _subsequently_ figure out which processes were started?
For debugging you would then attach to those processes in "hindsight". Ok, I already see the limitation of this approach: You cannot debug a node easily from the very beginning...

ooeygui, 2020-08-08:
@rhaschke That's correct. Without changes to roslaunch to output the command line for each executable, we can't know what roslaunch intended to do. (Currently the VSCode ROS2 module doesn't support launch debugging; modern roslaunch does not dump launch files in a way that we can hook. This is still a head-scratcher...)

VSCode C++ and Python extensions both support post-launch attach. 
This means you can start a ROS graph and attach to it. The VSCode roslaunch debugging subsystem is a helper to bootstrap debugging sub-graphs.

gautz, 2020-08-27:
It would indeed be a great feature for beginners in MoveIt. It would e.g. allow to debug [MoveIt tutorials](https://github.com/ros-planning/moveit_tutorials/). I suggest to explain this limitation in your doc, [e.g. here](https://github.com/ms-iot/vscode-ros/blob/master/doc/debug-support.md#launch). Namely: "debugging launch files running RVIZ is not supported yet."