ms-iot/vscode-ros #233: [bug]

thunderbug1, 2020-06-10:
**what is the bug**
I am trying to install the extension in an ubuntu docker image like so:

# install vs code server
RUN curl -fsSL https://code-server.dev/install.sh | sh

# Install vs code extensions
#ADD ressources/vs_code_extensions  /home/grip/.vscode/extensions
RUN code-server \
   --install-extension ms-vscode.cpptools \
   --install-extension eamodio.gitlens \
   --install-extension ms-python.python \
   --install-extension njpwerner.autodocstring \
   --install-extension streetsidesoftware.code-spell-checker

but this extension has a problem:

info  Wrote default config file to ~/.config/code-server/config.yaml
info  Using config file ~/.config/code-server/config.yaml
Installing extensions...
Installing extension 'ms-vscode.cpptools' v0.28.2...
Extension 'ms-vscode.cpptools' v0.28.2 was successfully installed.
Extension 'ms-iot.vscode-ros' not found.
Make sure you use the full extension ID, including the publisher, e.g.: ms-dotnettools.csharp
Installing extension 'eamodio.gitlens' v10.2.1...
Extension 'eamodio.gitlens' v10.2.1 was successfully installed.
Installing extension 'ms-python.python' v2020.5.2...
Extension 'ms-python.python' v2020.5.2 was successfully installed.
Installing extension 'njpwerner.autodocstring' v0.4.1...
Extension 'njpwerner.autodocstring' v0.4.1 was successfully installed.
Installing extension 'streetsidesoftware.code-spell-checker' v1.8.0...
Extension 'streetsidesoftware.code-spell-checker' v1.8.0 was successfully installed.

ooeygui, 2020-08-08:
Changed the name. I'm not sure why it wouldn't find it.

thunderbug1, 2020-08-10:
The problem affects code-server as well as vs-code.
I am behind a company proxy and my colleagues have the same setup  but no issues.
When I use a hotspot to connect to the internet it works without any issues.
I already tried to find possible issues with our IT but they say that everything is configured the same for my laptop as for everyone else.
The problem is that I don't even know which IPs should be reachable for everything to work correctly.
I tried to find it out with wireshark but that didn't work either.

ooeygui, 2020-08-10:
@thunderbug1 It sounds like this problem isn't isolated to the vscode-ros extension. I was considering that maybe this was due to a missing dependency - the vscode-ros extension requires both python and cpptools.

ooeygui, 2020-08-10:
Are you able to install it manually afterward?

ooeygui, 2021-12-27:
Closing due to external product.

closed 2021-12-27 by ooeygui