ms-iot/vscode-ros #239: [bug] URDF Preview renders only last visual, however should all

anton-matosov, 2020-06-15:
**general information**
Pop!OS 20.04 (based on Ubuntu)
extension v0.6.3
vscode 1.46.0

**what is the bug**
URDF Preview renders only the last <visual /> element of the link, which is incorrect. Rviz renders all
[image]

**expected behavior**
All <visual /> elements should be rendered. 
Official docs http://wiki.ros.org/urdf/XML/link mention 
>Note: multiple instances of <visual> tags can exist for the same link. The union of the geometry they define forms the visual representation of the link.

Rviz rendering
[image]

**additional context**
```
<?xml version="1.0"?>
<robot name="dd_robot">
    <link name="base_link">
        <visual name="body">
            <origin xyz="0.0 0.0 0.0" rpy="0.0 0.0 0.0"/>
            <geometry>
                <box size="0.5 0.5 0.25"/>
            </geometry>
        </visual>
        <visual name="caster">
            <origin xyz="0.2 0.0 -0.125" rpy="0.0 0.0 0.0"/>
            <geometry>
                <sphere radius=".05"/>
            </geometry>
        </visual>
    </link>
</robot>
```

seanyen, 2020-06-18:
@anton-matosov Thanks for the reporting. We will be looking into that soon.

ooeygui, 2023-05-04:
Corrected in new URDF previewer

closed 2023-05-04 by ooeygui