ms-iot/vscode-ros #240: [feature] Print pretty when c++ ros program is launched thru attach or launch

lusu8892, 2020-06-18:
**what is needed**
May I ask how to setup pretty-printing when a c++ ros node is launched thru

type: ros
request: attach

or 

type: ros
request: launch

The pretty-printing setting for gdb can be easily setup thru setupCommands when launch a C++ program of type: cppdbg. But this setupCommands seems not supported by any type: ros program.

seanyen, 2020-06-18:
@lusu8892 Thanks for the feedback. And sorry I am not familiar with the `pretty-printing` thing, can you link some documents, maybe write down steps (or screenshots) to help the team understand the scenario better?

lusu8892, 2020-06-18:
[image]

So, as shown in the pc, there are two programs defined in launch.json, only the first could turn on pretty-printing for the underlying gdb debugger. But for the second case, the setupCommands can not be mentioned, so pretty-printing can not be turned on. Pretty-printing is a feature from GDB debugger. Here is [link](https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_57.html)

danneboom, 2021-06-21:
I'd really appreciate this feature. It enables you to debug std library containers like strings, vectors, maps etc. Without this line, the variables view is simply not usable.

Just check this example:
````cpp
 std::vector<double> test = {1, 2, 3, 4, 5};
````

Without pretty printing:
[image]

With pretty printing enabled:
[image]

Please enable pretty printing - its a must have.

fmros, 2022-01-31:
You can still use the "standard" gdb-attach approach which works for me and makes pretty printing work from attach:
```
{
            "name": "(gdb) Attach",
            "type": "cppdbg",
            "request": "attach",
            "program": "${workspaceFolder}/devel/lib/ros_package/ros_package_executeable",
            "processId": "${command:pickProcess}",
            "MIMode": "gdb",
            "setupCommands": [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                }
            ]
        },
```

ooeygui, 2022-10-13:
We have added pretty printing to the default launch.

closed 2022-10-13 by ooeygui