ms-iot/vscode-ros #243: [bug] catkin build tasks not working

PeterMitrano, 2020-06-24:
**general information**
Ubuntu
0.6.3

**what is the bug**
Trying to run build gives me 
```
The task provider for "catkin_cmake" tasks unexpectedly provided a task of type "catkin_make".
The task provider for "catkin_cmake_isolated" tasks unexpectedly provided a task of type "catkin_make_isolated".
```
And the terminal appears to do nothing, showing only:
```
> Executing task: catkin_make --directory "/home/pmitrano/catkin_ws" <
```

**expected behavior**
catkin workspace builds

**additional context**

PeterMitrano, 2020-06-24:
It seems something isn't configured right, since none of the ros features seem to work. Trying to start ros core says it failed to launch (not sure where to look for error information?).

PeterMitrano, 2020-06-25:
I fixed this using this answer:

https://stackoverflow.com/questions/54092486/visual-studio-code-terminal-blank-screen

and also by fixing my environment variables

closed 2020-06-25 by PeterMitrano

seanyen, 2020-06-26:
cc @humanoid2050, does this look like the same issue you are trying to solve by #241?

humanoid2050, 2020-06-26:
Same error I had.

PeterMitrano, 2020-06-26:
just to clarify, I still see this message printed -- but the catkin_make works

humanoid2050, 2020-06-26:
I could run the catkin_make only so long as I didn't try to make it there default build task. Making it default caused the error for me.