ms-iot/vscode-ros #246: Debugging large projects from launch file

Kuppharish, 2020-06-25:
I have been using a SLAM package called lego loam. I have followed all the steps to setup the configuration and debug the launch file. When I run debug (or hit F5). The launch file runs without any errors but only few nodes are active. In the rqt_graph I can see that many nodes and topics are **inactive**. When I run launch directly from terminal I can see that all the nodes are active. 

It is not a problem of debug because even when I try **run without debugging**, all the nodes are not getting active.

seanyen, 2020-06-26:
@Kuppharish Hi thanks for reporting. Can you share your SLAM package or your roslaunch file?

Kuppharish, 2020-07-01:
Hey @seanyen this is the link of the package I am using,
https://github.com/RobustFieldAutonomyLab/LeGO-LOAM

ooeygui, 2022-10-13:
We have added support for just attaching to specific nodes. Checkout `attachDebugger` in the documentation.

closed 2022-10-13 by ooeygui