ms-iot/vscode-ros #253: [bug] URDF preview not displayed correctly when using too many spaces in joint origin xyz or rpy

Tuebel, 2020-07-08:
**general information**
\<OS version, Ubuntu 18.04, ROS Melodic\>
\<Version of the plugin 0.6.3\>

**what is the bug**
When specifiying the following as origin of one of my joints, some parts of the URDF are not displayed. Note the second space between ${PI / 2} and  0 in rpy.
`<origin xyz="0  0 0.2025" rpy="${PI / 2}  0 ${PI}"/>`
[image]

Doing the same for xyz yields the following. Note the second space between 0 and 0.
`<origin xyz="0  0 0.2025" rpy="${PI / 2} 0 ${PI}"/>`
[image]

**expected behavior**
When removing the additional spaces and using only one space between each value I get the correct preview:

[image]
`<origin xyz="0 0 0.2025" rpy="${PI / 2} 0 ${PI}"/>`

**additional context**
Uploading the xacro to the robot_description parameter results in a correct display in RViz no matter how many spaces I use between the values.

ooeygui, 2020-08-08:
Internally, the extension uses RosWebTools to render. Looking at the RosWebTools, I'm not sure why this manifests - it seems to do the right thing.

ooeygui, 2023-05-04:
Corrected in new URDF Previewer

closed 2023-05-04 by ooeygui