ms-iot/vscode-ros #254: [bug] There are task errors. See the output for details.

robokamran, 2020-07-08:
**vscode 1.46.1**
I have vscode for compiling ROS projects on  WSL2 with ubuntu 18.04. I installed the new ROS plugin, project is configured correctly but on trying to build the project with `ctrl+shift+b` I get this pop-up and the following messages on the output window. The making is performed correctly, what is this message trying to say??

	The task provider for "catkin_cmake" tasks unexpectedly provided a task of type "catkin_make".
	The task provider for "catkin_cmake_isolated" tasks unexpectedly provided a task of type "catkin_make_isolated".

[image]

seanyen, 2020-08-04:
This should be fixed in the latest release 0.6.4. Please upgrade the extension and reopen if you still hit the issue.

closed 2020-08-04 by seanyen

robokamran, 2020-08-05:
that's fixed thanks!