ms-iot/vscode-ros #260: [bug] Could not source the setup file for ROS distro

cwillia109, 2020-07-14:
**general information**
OS: Ubuntu 18.04
Plugin: 0.6.3
VS Code: 1.47.0
ROS Distro: Melodic

**what is the bug**
Using a fresh install of visual studio code, the extension gives a "Could not source the setup file for ROS distro '/opt/ros/melodic/setup.bash'"

**expected behavior**
The plugin is able to source the ROS setup.

**additional context**
Have tried this with the previous 2 versions of the plugin, neither worked. Python is also able to get ROS related autocomplete, but not C++.

PeripateticWind, 2020-07-21:
My environment is the same as yours, and encountered the same problem.

nightduck, 2020-10-03:
The user-fix I found is to select the "workspace" tab in the extension's settings and type "melodic" into the distro box.
[image]

BlueBirdHouse, 2022-02-22:
I do this. However, there is no effect.

[image]
[image]

ooeygui, 2022-02-22:
I'm adding support for specifying a startup script, both in the workspace and globally - https://github.com/ms-iot/vscode-ros/pull/678

ooeygui, 2022-02-22:
To add C++ intellisense, you need to use the `ROS: Update C++ Properties` pallet command. It is not done automatically since it will overwrite existing data.

ooeygui, 2022-10-13:
Added support for custom scripts.

closed 2022-10-13 by ooeygui