ms-iot/vscode-ros #261: [bug] URDF preview not displaying meshes in vscode 1.47

Tuebel, 2020-07-14:
**general information**
Ubuntu 18.04
Plugin version 0.6.3
vscode 1.47

**what is the bug**
In vscode 1.47, meshes do not get loaded / displayed for my URDF. Simple geometries like cylinders get displayed correctly.
[image]

**expected behavior**
Expected behaviour works in vscode 1.46. My robot model gets displayed after a few tries (see #93).
[image]

**additional context**
-

closed 2020-07-15 by ooeygui

gonultasbu, 2021-12-09:
Hello, is this issue fixed? It seems like I am still experiencing the same issue.

ooeygui, 2021-12-09:
Hi @gonultasbu ,
This specific issue has been corrected and verified working with 0.6.9 and vscode 1.63.0.
Can you share your vscode data? In VSCode, goto `Help`, `About`, then select copy, and paste it as a comment? 

Internally, the VSCode extension uses a webview which hosts ROS Web Tools. 
Are you disconnected, or on a network which requires a VPN or Proxy?

gonultasbu, 2021-12-09:
Hi @ooeygui, thank you for your quick response.

I am running ros and vs code in WSL2, maybe that could be the reason?

`Version: 1.63.0 (user setup)
Commit: 7db1a2b88f7557e0a43fec75b6ba7e50b3e9f77e
Date: 2021-12-07T06:26:56.179Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.22000`

ooeygui, 2021-12-09:
Thanks for clarifying, and working with me to diagnose this. I'm not able to repro this using the WSL remote extension. 
Are you running VSCode in WSL using the desktop rendering bridge?

Can you do the following:
* Close the blank URDF preview. 
* Goto `Help` `toggle Developer tools`
* Then reopen the URDF preview on the URDF file

Can you share the output you see in the debug tools console window?

gonultasbu, 2021-12-10:
Thanks for your help! This is the error I get.  

`Could not load geometry mesh: ./meshes/mesh_cylinder.obj`

The referred geometry mesh is the file I have in the corresponding folder, which is read correctly by the simulator I am using. Maybe I need to state the full path to the .obj mesh I am using and relative paths don't work?

ooeygui, 2021-12-10:
Can you try prefixing the mesh object `package://<packagename>/meshes/mesh_cylinder.obj` in the URDF?

In order to work within the VSCode sandbox, we have to modify the paths that is encoded in the urdf. it doesn't currently handle relative paths. I created https://github.com/ms-iot/vscode-ros/issues/623 to track it. Would it be possible for you to add an example to that new bug?

gonultasbu, 2021-12-10:
I tried giving absolute system file path like this:

`/mnt/c/Users/burak/Documents/GitHub/diffnav/data/urdf/simple_ackermann/meshes/mesh_cylinder.obj` or 

`file:///mnt/c/Users/burak/Documents/GitHub/diffnav/data/urdf/simple_ackermann/meshes/mesh_cylinder.obj`
 
but it still returns the same could not load geometry mesh error. Would that be also considered under the same bug?

ooeygui, 2021-12-10:
Unfortunately, yes, it's the same bug. ultimately the code tries to replace package://<packagename> with a vscode redirect. 
Here's the codepath that is failing for fully qualified or relative paths - https://github.com/ms-iot/vscode-ros/blob/691323d5c4ec429977e8eab85c094612b47060e4/src/urdfPreview/preview.ts#L119.

gonultasbu, 2021-12-10:
Thanks for the guidance. Is there a workaround for this issue for the current version?  A sample working path string would be very helpful here.

ooeygui, 2021-12-10:
The workaround is to refer to the mesh using the ROS package:// URI. The mesh needs to be in a ROS package, and the URI would include the package name. for example: if I have a ROS package called `foo`, with a mesh folder in it containing a file `mesh.dae`, I the URDF would link to `package://foo/mesh/mesh.dae`. 

Fixing this is going to take a bit of time; I need to release 0.7.0, then I can include a fix in 0.8.0.

gonultasbu, 2021-12-10:
Thanks for your assistance; I will comment some examples to #623 tomorrow.

PR #263 (2020-07-15): updated to use vscode-webview-resource for vscode 1.47
Since vscode 1.47, the usage of `vscode-resource` will be generating errors and it recommended to migrating to `vscode-webview-resource`. And also now it is recommended to use `https` to access the external resource.

Fixes #261 and #214.