ms-iot/vscode-ros #288: [feature] Intellisense for .msg files

Timple, 2020-08-27:
**what is needed**
Intellisense for .msg files

**why is it needed**
Often messages consists of other message types. Take for example the [odometry](http://docs.ros.org/melodic/api/nav_msgs/html/msg/Odometry.html) message.

Allowing ctrl+click on these types and ending up in that message would help the workflow a lot.

martinerk0, 2021-07-10:
Shouldn't intellisense somehow do autocomplete for msgs since they are compiled into C++ and Python classes? I too have problems that in some functions with some messages autocomplete does work and with other does not. It Would be cool if someone provided **how to setup C++ extension/intellisense/autocomplete** in  ROS since there are ton of threads/questions on the internet and lot of people have problems with it.

ooeygui, 2021-07-13:
When there is no c_cpp_properties.json file in the workspace, the VSCode ROS extension synthesizes one from the environment, which includes the AMENT/Catkin environment, development workspace, and ROS workspace includes. This works for msg files as well:
[image]

If you have an existing configuration, you can use `ctrl` + `shift` + `p` to bring up the command pallet, and type `update` to find `ROS: Update C++ properties` or `ROS: Update Python Path` to adjust the intellisense:
[image]

If this doesn't work, I'd love to understand more about your environment.

gavanderhoorn, 2021-07-13:
I believe the request is to be able to auto-complete etc actual `.msg` files. Not the generated C++ sources.

ooeygui, 2021-07-13:
@gavanderhoorn Ah - thanks for clarifying. Reopened.

gavanderhoorn, 2021-07-13:
Would that need a language server implementation for the [msg](http://wiki.ros.org/msg) format?

For ROS 2, this could simply be DDS OMG IDL.

ooeygui, 2021-07-13:
I'm going to have to dig into this. https://code.visualstudio.com/api/language-extensions/programmatic-language-features. I think you are right, it would need a language server and symbol database.

gavanderhoorn, 2021-07-13:
Could be a nice internship project. The `msg` spec isn't that complex and frameworks like [openlawlibrary/pygls](https://github.com/openlawlibrary/pygls) should reduce complexity even more (as it takes care of the protocol layer).

OMG IDL though .. :)

OsamaAlAhmadieh, 2023-02-17:
Not sure about the progress on this. One thing that didn't work for me is also intellisense highlighting and "go to definition" in other files in the same workspace for the self defined messages. I had to manually edit the `python.analysis.extraPaths` in order to get it working. Is it something that should also be automatically configured within vscode when I define a new library or it's not implemented so? Also, after adding a new message type and building again, the intellisense highlighting didn't work until after I updated my python path for ROS in Vscode. 

Specifically, I was working with ROS2 foxy, on ubuntu 20.04 using vscode version 1.75.1. and my file structure looked as so. 

- ros2_ws/
  -  src/
      - package1/
        - package 1/  
          - script 1
      - package2/
        - msg/    
          - my_msg1.msg

apologies for the noob representation of the folder structure. 
but my msg was defined in the `package2/msg/my_msg1.msg` and I was importing it to be used in the node defined in `package1/package1/script1` and when I added for example a new `my_msg2.msg` that's when I had to updated the ros python path in vscode. 

I hope this helps!