ms-iot/vscode-ros #299: Python Unresolved Import Error for Catkin Workspace[bug]

cmuell89, 2020-09-15:
**general information**
Linux Mint 19.3
ROS Melodic
ROS Extension v0.6.4

**what is the bug**
None of my locally developed packages built using catkin_make/catkin build are successfully resolved as imports despite the inclusion of python.autoComplete.extraPaths. 

**expected behavior**
I would hope to have my local ROS packages incorporated into the import tree for happy development. 

**additional context**

Tested with jedi, Microsoft, and Pylance Language Servers. 

The update path commands seem to updated my settings.json and c_cpp_properties.json files withing .vscode directory. I have it setup such that .vscode exists at the sample level of src/ in my catkin workspace. 

settings.json
```
{
    "python.autoComplete.extraPaths": [
        "/home/carl/cairo/sawyer_ws/devel/lib/python2.7/dist-packages",
        "/opt/ros/melodic/lib/python2.7/dist-packages"
    ],
    "ros.distro": "melodic",
    "python.pythonPath": "/usr/bin/python",
    "python.linting.pylintEnabled": false,
    "python.linting.enabled": true,
    "python.linting.flake8Enabled": true
}
```

cmuell89, 2020-09-15:
Could be related to:
https://github.com/microsoft/vscode-python/issues/13357

cmuell89, 2020-10-01:
Update. This is not working with the Jedi server. Jedi appears to parse one file at a time. So when I open an individual file, then the import errors appear.

Anyone else having issues with this?

cmuell89, 2020-10-01:
Many users of ROS still use Python 2.7. Pylance Language Server does not support Python 2.7 syntax and the linking and path updates inside the __init__.py of your built ROS packages in devel/lib/python2.7/dist-packages directory fails to parse correctly.

I think it would be worth documenting this. Switching to Jedi language server works correctly.

TobiMiller, 2020-10-16:
How did you fix it? 

In Settings (UI) i have set the language server to Jedi. Thats it? Doesnt work for me.

cmuell89, 2020-10-29:
Yea Jedi seems to work for me out of the box. Not sure what else to recommend.