ms-iot/vscode-ros #305: [enhancement] Allow customization of the provided tasks

anton-matosov, 2020-09-21:
**what is needed**
vscode-ros extension provides a set of catkin/catkin_make/colcon tasks for building and running tests. However these is no way to customize them right now and developers need to re-create setup from scratch.

**why is it needed**
1. Create a debug build by providing `-DCMAKE_BUILD_TYPE=Debug` to `catkin_make: build task`
2. Run specific tests by customizing run_tests command as described [here](http://docs.ros.org/jade/api/catkin/html/howto/format2/run_tests.html)

anton-matosov, 2020-09-21:
I have start working on this in this branch https://github.com/anton-matosov/vscode-ros/tree/catkinArgs

anton-matosov, 2020-09-21:
After working with this for a little while, I realized that all the tasks can essentially be implemented as a single generic task:
`ros-shell` - provides `vscode.ShellExecution` with `env: extension.env` and takes `command` and `args` options
That approach will simplify existing implementation and will provide useful utility to developers allowing them to run any ROS command in the shell environment provided by extension

ooeygui, 2021-12-27:
Forgot to close this after the merge. Thanks for the contribution!

closed 2021-12-27 by ooeygui