ms-iot/vscode-ros #309: Debug - connect to move group

myFree, 2020-09-26:
Thanks for this fantastic plug-in. This is probably more a how-to question than an issue; so please let me know if there is a better channel to post it on.

**what is needed**
I would like to connect a debugger to the running move_group node; in order to debug and step through the source code

**why is it needed**
In line with the great debugging support the plug-in already offers; this can be incredibly helpful for learning code sets

I'm running ROS inside a docker container; and connecting vs-code from my host via the process described in https://code.visualstudio.com/docs/remote/attach-container

I'm able to connect to and debug a custom node I am writing via https://github.com/ms-iot/vscode-ros/blob/master/doc/debug-support.md#launch-and-debug-python-and-c-nodes; so I've got some of the setup correct.

I've started up a complex .launch file directly in the container and would like to attach to the `/move_group`. I've also installed `ros-melodic-moveit-core-dbgsym` which as I understand should give debug symbols. However when I create a ROS-Attach debug configuration and run it; I see the blue progress bar for a few seconds and then nothing happens.

I don't see any output in the vs-code debug terminal either.

Any suggestions on how I can trouble shoot this further would be most welcome. Or if I need to try a different approach - I'm open to all suggestions.

Thanks again for the great plug in.

myFree, 2020-09-26:
I have not build moveit from source; and am just opening the repo which I've cloned. I was wondering if I would be able to debug only by installing the debug symbols package - `melodic-moveit-core-dbgsym`

If I run a Ros Debug connect config in the vs-code which has my custom ROS node which I have built; then it successfully goes to the next step of letting me select a process. But; I don't have the moveit_core sources open; so I'm not sure how I would set break points.

ooeygui, 2020-09-26:
@myFree Thanks for the report and the feedback! 
Debugging in a container (and WSL2) is something we want to support better, and your write up gives us something to work toward. 

Just so I can reproduced your scenario, can you document how you set up ROS and MoveIt in the container and about the container host you are using? (If you are following instructions for this specific setup, if you want to point to docs that works too). Just want to make sure I'm using the same configuration.

Thank you!

myFree, 2020-09-27:
My host OS is Ubuntu 20; and I have Docker version 19.03.12 installed.

I do not currently have a docker file I can share; as I've been learning ROS through this process; so have been taking frequent "commits" as I build up my environment. If this becomes needed; I can work on creating one.

If there are any specific details from the docker image that could be of higher interest please let me know so I can share them.

In terms of trying to debug it further from my side - are there any settings I can change to get more granular logging? Currently I don't see anything in the Terminal | OUTPUT | select ROS from the drop down.

I have played around with vs-code extensions a little bit before; but I don't know if its possible to debug an extension when the extension is being run within a docker container?

Sorry if I'm not providing much useful info to go on; but do let me know and I will try gather the relevant data

myFree, 2020-09-28:
After building moveit from source; and am then able to attach to the move_group node and hit break points

So all of the above issues must be with my setup and trying to use `melodic-moveit-core-dbgsym`

closed 2020-09-28 by myFree

ooeygui, 2020-09-28:
Thank you for confirming. Be Well!