ms-iot/vscode-ros #314: Red squiggly lines in code that compiles correctly: name followed by '::' must be a class or namespace name

myFree, 2020-10-07:
Happy to copy across the full details if that is helpful; but it seems that the default c++ support is understandably not aware of all the intricacies of the ROS build process (which is fairly understandable).

Do you have any suggestions on if it would be possible to address this?
https://github.com/microsoft/vscode-cpptools/issues/6252

JanezCim, 2020-10-19:
Red squiggly lines for `ROS_INFO` and for `ros::Time()` dissapeard for me when i switched `"cppStandard": "gnu++14"` to `"cppStandard": "c++14"` in the `.vscode/c_cpp_properties.json`

source of solution: https://github.com/microsoft/vscode-cpptools/issues/5963#issuecomment-685010371

myFree, 2020-10-19:
WONDERFUL; thank you!
Reallly helps spot actual issues when every line of logging isn't being incorrectly flagged.

closed 2020-10-19 by myFree