ms-iot/vscode-ros #315: command 'ros.previewUrdf' not found [bug]

droter, 2020-10-07:
**general information**
Ubuntu 16.04
0.6.4

**what is the bug**
Error:  command 'ros.previewUrdf' not found

Steps to reproduce:
ctrl+p
search for ros
click ros.previewUrdf

**expected behavior**
Open preview window with robot model

**additional context**
Used to work.  Not sure what has changed.

Full Error Message:
Command 'ROS: Preview URDF' resulted in an error (command 'ros.previewUrdf' not found)

droter, 2020-10-26:
I uninstalled code and reinstalled.  Fixed the problem.

ooeygui, 2020-10-26:
@droter Thanks for the report. Sorry I missed this earlier.  The URDF Preview requires a ROS workspace be detected, in order to source the environment that should be used to load the URDF. If the extension does not detect the environment, the commands do not work. I'm investigating a way to make some commands like URDF work independent of a workspace, but it would still require the ROS environment to be sourced in order to find the tools.

kvnptl, 2021-02-08:
> @droter Thanks for the report. Sorry I missed this earlier. The URDF Preview requires a ROS workspace be detected, in order to source the environment that should be used to load the URDF. If the [...]

I am facing the same issue. Can you please help me with how to setup?

I am using Windows 10 and VS Code 1.53. I have installed ROS extension as well.

ooeygui, 2021-02-08:
@kvnptl I hope this helps:
The VSCode extension leverages the ROS environment from your terminal window in order to find tools like Xacro.

For those who might not know - A catkin workspace is like a visual studio solution, with a specific layout of directories. The "workspace directory" has a "src" directory inside of it, which contains the packages that make up your robotics project. When you build with catkin, it creates a file `.catkin_workspace`, which is what this extension keys off of. (Otherwise loading the extension would be very expensive, we only do the expensive initialization with this key).

Before loading VSCode:
  * Open a terminal window and source the ros environment by ensuring you've run c:\opt\ros\noetic\setup.bat.
  * cd into your catkin workspace. 
  * Build yours solution from the command line using `catkin_make`.
  * From the root of the workspace open `vscode .` so it loads with the root of your workspace.

After the extensions load (which can take a few seconds to a few minutes in vscode, depending on the number of extensions installed), you should be able to open a URDF.

VSCode isn't as self contained as Visual Studio. It does have to inherit environment for building and loading packages. We have some helpers to try to source the environment, but it does work best when bootstrapping like above.

nicolaje, 2021-09-02:
Hi @ooeygui ,

I am using VSCode inside a Docker container with the Remote containers extension. 

Everything is setup so that VScode opens the ROS workspace.

Sometimes the ROS extension works perfectly, and sometimes it doesn't work, see the following message :

[image]

I can't explain the reason of why it suddenly doesnt work.

Do you have any input or ideas ?

ooeygui, 2021-09-02:
@nicolaje Thanks for the report. The ROS extension is getting quite large, so can take time to load. I have a work item in 0.7.0 to use webpack (fixing https://github.com/ms-iot/vscode-ros/issues/357) to improve launch performance. 

To make sure that this isn't a perf bug, can you verify and answer a few things for me? 
Did you installed the ROS extension in the container?  (from the screenshot, it doesn't look like it; if you restarted the container, it defaults back to the original state - requiring a reinstall into the container)
If it is, did the ros distro get selected correctly?
Is there a ros install in /opt in the container?

nicolaje, 2021-09-07:
Hi @ooeygui Lou,

thank you for your feedback.

The extension is listed in the devcontainer.json "extensions" list and is correctly installed inside the container.

Following are the **weird** steps that got everything working again :

I put "/opt/ros/noetic/setup.zsh" in the Ros: distro settings of the ROS extension and rebuilt + reopen the container

That of course got me the error message that the setup file for "/opt/ros/noetic/setup.zsh" distribution can't be sourced **but** :
I noticed several extensions I have in my devcontainer.json extensions list appeared again in my bottom "task bar" (Jira, Bitbucket...), and I realized they were not activated before also, as can be seen in my previous screenshot.
I then put "noetic" only in the settings of the ROS Extension and now I have it activated again.

Conclusion, something was blocking the activation of several extensions including the ROS one, and editing the settings of the ROS one got the others to load again, + the ROS one.

Not sure that helps, but there seems to be something behind that, don't hesitate if you need more tests. This also should belong to a separate issue, the symptoms were looking similar.

ooeygui, 2021-09-07:
@nicolaje Thanks for the context. I'll do some digging. Some things stood out - including using zsh. (The extension is explicitly looking for bash)

samwesley, 2022-01-14:
Hey @ooeygui I'm also seeing this error, using ROS Kinetic on Ubuntu 16.04 on a Raspberry Pi, I've tried the above steps from Feb. 8th with no luck

ooeygui, 2022-01-14:
@samwesley Kinect is not supported by the ROS extension (Kinetic and Ubuntu 16.04 are both EOL).

The ROS extension uses ROS under the covers. It will only activate fully when it detects a catkin (Melodic or Noetic) or colcon (ROS2 Foxy+) workspace.

samwesley, 2022-01-14:
Ahh, thanks @ooeygui!