ms-iot/vscode-ros #325: ROS2 Daemon Start command not available

nightduck, 2020-10-23:
**General information**
Ubuntu 18.04 and ROS2 Dashing
vscode-rot v0.6.4

**What is the bug**
The ros2 commands referenced in PR #103 are not present. It's unclear if I'm using them correctly as the ROS2 functionality is so poorly documented.

**Expected behavior**
Type Ctrl+Shift+P, start typing "ros2" and have the available commands show up. I only see ROS commands.
[image]

**Additional context**
This little status icon in the lower bar correctly detects Dashing. I've also seen a check mark on occasion, and can open the Daemon status window when that's the case. So it's possible to accidentally turn the ROS2 Daemon on, but that happens very seldomly.
[image]

ooeygui, 2020-10-23:
Hi @nightduck ,
Thank for the report and screen shot.

The command names are not different between ROS1 and ROS2. After you select the command, we fork off between ROS1 and ROS2 depending on the detected environment for the workspace. (Either by inheriting it from a launched terminal, or by introspecting a previously built workspace).

We would definitely like to improve documentation, and are working on videos.  
Is there specific feedback we can address?

nightduck, 2020-10-23:
Thank you for the quick reply!

For starters, how would I intentionally start the ROS2 daemon?

Also, my ROS2 workspace doesn't seem to be sourced. Going off your comment, I should be able to type the rosrun command, and select my package and then executable launch file. My packages don't show up on the auto-complete list.
[image]
[image]

Additionally, how do I pass arguments to a colcon type build task? My project requires certain cmake args, so I've tried a standard shell type build task, but then it'll complain about how ament is not available (because that build task doesn't source dashing, presumably). My workaround is building the project in a terminal separate from VSCode.

Finally, issues with the debugging step. If I put a path to a ROS2 python launch file in my launch task and run that, VSCode will complain about needing an absolute path (which I gave it). I assume that's because it was expecting an xml file, and that's the generic error message it gives for IO Exceptions. I also understand that ROS2 in general has problems prefixing gdb to their python launch files, so if you blocked this feature intentionally, that would be understandable.

But if I use the ROS Attach feature, I think it manages to attach to my ROS2 node, but I never get a gdb prompt and it never hits any of my breakpoints. I get similar behaviour when trying to manually attach gdb outside of VSCode, so this is possibly a me-problem.

ooeygui, 2020-10-23:
Thank you for the writeup, I appreciate the details. (We have a separate documentation bug here https://github.com/ms-iot/vscode-ros/issues/205)

ROS2 debugging isn't well supported at this time. The model we used for ROS1 isn't present in ROS2. (Although Foxy may support the tooling we need in order to implement launch debugging, we have it on the backlog. 

Our ROS2 extension uses C++ and Python extensions under the covers; It's really just a front end. We have seen some issues with the Python extension & ROS (https://github.com/ms-iot/vscode-ros/issues/292, which may affect ROS2), which should be fixed shortly. 

There is a outstanding pull request (https://github.com/ms-iot/vscode-ros/pull/310) for builds. (I am working on running this through on all platforms to pull).

JaehyunShim, 2020-10-26:
@ooeygui 

Are you guys also planning to implement the ROS2 lint system as a part of the extension?

Jaehyun

ooeygui, 2020-10-26:
@JaehyunShim No, we do not have the lint system on the backlog. I agree - it would be great to have support. Please post an independent feature request on the Github issues.