ms-iot/vscode-ros #330: [feature] Enable Debugging rostests

ooeygui, 2020-10-27:
**what is needed**
Currently launch debugging works on ROS1, but does not allow debugging rostest.

**why is it needed**
Because tests...

RyanDraves, 2021-04-14:
Could you describe more details on what needs to be added with this feature request? I'm working on general `rostest` support, but I've noticed that I'm already able to the existing launch debugging on rostest files. `roslaunch --nodes [launch file]` and `roslaunch --args [node] [launch file]` is already doing a good job parsing rostest syntax (`<test/>` over `<node/>`), so test nodes are successfully being launched in the debugger, even if `roslaunch my_package my_test_file` would fail. I'm using ROS Noetic to test it, so it is possible that `roslanch --nodes` parsing changed recently.

ooeygui, 2021-04-14:
Hi @RyanDraves,
Thanks for the note. The VSCode Debugger needs to launch the executables under the appropriate debug subsystem. It uses `roslaunch --dump-params` to introspect the parameters that roslaunch would have launched each executable with, then launches them independently with the debugger. roslaunch with a rostest file actually fails in this codepath, so we can't get the launch parameters.

I believe we would need to modify roslaunch, or use the roslaunch python APIs to process launch files directly.

RyanDraves, 2021-04-14:
Thank you for the quick reply! Could you expound on situations in which `roslaunch --dump-params` is failing? I added parameters to my rostest file, as well as test cases to assert their values, and everything is working for me just fine.

Here's the rostest file I created:
```
<launch>
    <param name="param1" value="3.5" />
    <test test-name="test_talker" pkg="test_package" type="test_package_test_talker">
        <param name="param2" value="param2" />
    </test>
    <test test-name="test_listener" pkg="test_package" type="test_package_test_listener">
        <param name="param2" value="param2" />
    </test>
</launch>
```

And here's the result I received when running `roslaunch --dump-params src/test_package/launch/test.launch`:
```
/param1: 3.5
/test_listener/param2: param2
/test_talker/param2: param2
```

Everything appears to be in order from me when using the launch debugger on a rostest file, although I haven't gotten to Python nodes yet as I'm afraid I'll run into #292 for ROS Noetic.

From some quick testing I did, it appears `roslaunch` will always parse <test> nodes, even if it's ignoring them when launching things. It'll even set the parameters embedded in <test> tags when not launching the nodes themselves.

ooeygui, 2021-04-14:
Thanks for the writeup. I'll see if I was hitting a transient bug.
I'll take a look at this again. 

Thanks for pointing out the other issue - that was fixed by our friends in the Python VSCode extension team. We were waiting for a deployment before closing that - I simply forgot :-(

PR #435 (2021-06-07): Add .test files to launch debugging (ROS1)
Following up on my conversation with @ooeygui, I reproduced the issues with attempting to use a `.test` file for launch debugging. This change allows a user to specify a `.test` file when creating a `ROS: Launch` entry in their `launch.json`, which should now hopefully resolve #330.

I'm unsure on any documentation updates that may be desired, so for now I left that untouched.
closed 2021-06-07 by ooeygui