ms-iot/vscode-ros #334: [bug] Unable to launch the ros2 launch file

sachinkum0009, 2020-11-06:
**general information**
\<OS version: 20.04, Ubuntu\>
\<0.6.4\>

**what is the bug**
\<unable to launch the ros2 launch file\>

when I use the roslaunch command from command pallete

```bash
ros2 launch ./zzzzz/ros2/basics/colcon_ws2/install/four_wheel_robot/share/four_wheel_robot/launch/empty_world.launch.py
```
this seems to be working
```bash
ros2 launch ~/zzzzz/ros2/basics/colcon_ws2/install/four_wheel_robot/share/four_wheel_robot/launch/empty_world.launch.py
```

Any idea how can we change this???

[image]
[image]
[image]
[image]
[image]

seanyen, 2021-01-20:
@sachinkum0009 Can you confirm if it is still a problem on `0.6.6`? We changed\fixed the logic on what to construct the full path for the launch file.

sachinkum0009, 2021-01-24:
Ok
Thank you 
👍

closed 2021-02-19 by sachinkum0009