ms-iot/vscode-ros #336: [bug] Intellisense/Autocompletion [C++] is not working!

milan-r-shah, 2020-11-06:
**general information**
OS version: Ubuntu 18.04 LTS
Version of the plugin: v0.6.4

**what is the bug**
Intellisense/Autocompletion is not working!

I have already tried configuring my `c_cpp_properties.json` through this extension by running `ROS: Update C++ properties` command. But still, IntelliSense is not working. Here is my `c_cpp_properties.json` file:
```
{
    "configurations": [
        {
            "browse": {
                "databaseFilename": "",
                "limitSymbolsToIncludedHeaders": true
            },
            "includePath": [
                "/opt/ros/melodic/include/**",
                "/home/milan/catkin_ws/src/my_ros_package/include/**",
                "/usr/include/**"
            ],
            "name": "ROS",
            "intelliSenseMode": "gcc-x64",
            "compilerPath": "/usr/bin/gcc",
            "cStandard": "gnu11",
            "cppStandard": "gnu++14"
        }
    ],
    "version": 4
}
```

**expected behavior**
Autocompletion should work!

**additional context**
In the extension setting: 
`Ros:`**`Distro`** 
`ROS installation distro to be sourced.`, 
I tried putting:

- `Melodic`
but it shows: `Could not source the setup file for ROS distro "Melodic".`

- `/opt/ros/melodic/setup.bash`
but it shows: `Could not source the setup file for ROS distro "/opt/ros/melodic/setup.bash".`

- even my local setup.bash i.e. `~/catkin_ws/devel/setup.bash`
but it shows: `Could not source the setup file for ROS distro "~/catkin_ws/devel/setup.bash".`

LucasWaelti, 2020-12-08:
Try to change from `"cppStandard": "gnu++14"` to `"cppStandard": "c++14"`. ROS is using the [c++14 standard](https://www.ros.org/reps/rep-0003.html#melodic), so specifying `gnu++14` seems to break things. This was deduced from [this question](https://answers.ros.org/question/341313/what-happens-if-i-use-c-17-features-in-my-ros-nodes/).

ooeygui, 2020-12-08:
@LucasWaelti Thanks for the investigation!

tranvnhan, 2021-01-04:
> Try to change from `"cppStandard": "gnu++14"` to `"cppStandard": "c++14"`. ROS is using the [c++14 standard](https://www.ros.org/reps/rep-0003.html#melodic), so specifying `gnu++14` seems to break [...]

It works for me. Thanks.

closed 2021-12-27 by ooeygui