ms-iot/vscode-ros #344: [bug] Wrong build tool detected

stertingen, 2020-12-01:
- [x] Windows: Windows 10 Enterprise Version 1909
- [x] Linux: Ubuntu 20.04.1 LTS (WSL) (uname -a: `Linux *** 4.4.0-18362-Microsoft #1049-Microsoft Thu Aug 14 12:01:00 PST 2020 x86_64 x86_64 x86_64 GNU/Linux`)
- [x] ROS 1: Noetic
- [ ] ROS 2:

Version of the plugin: v0.6.4

Copy the Version information from the `Help | About` menu:
```
Version: 1.49.1 (system setup)
Commit: 58bb7b2331731bf72587010e943852e13e6fd3cf
Date: 2020-09-16T23:27:51.792Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Windows_NT x64 10.0.18363
```

# what is the bug
Under some circumstances, the ROS extensions selects the catkin_make build tools when catkin tools would be correct.

## Repro steps
1. Create ROS workspace
2. Build with `catkin_make`
3. Remove `build` and `devel` directories
4. Build with `catkin build`
5. Load ROS workspace in VS Code
6. Open build tasks (Ctrl+Shift+B) and see `catkin_make` and `catkin_make_isolated` as if the workspace was recently build with `catkin_make` (obviously, `catkin_make` fails because the workspace was build with `catkin build`)

# expected behavior
See `catkin` as build task

# additional context
Currently, the extension detects the build tool by looking if the file `.catkin_workspace` (placed by catkin_make) or the directory `.catkin_tools` (placed by catkin build) exist. Both tools also place a file at `devel/.built_by` which contains the name of the build tool (`catkin build` or `catkin_make`).

(Bonus: Right now the names of the build tasks are very confusing. `catkin_make: build` executes `catkin_make`, `catkin: make` executes `catkin build`. The build task for `catkin build` should be `catkin: build`, `catkin tools: build` or `catkin tools: make`.)

ooeygui, 2020-12-03:
Thanks for the report. Today the extension depends on the behavior of catkin_make in order to find the environment.

stertingen, 2020-12-04:
> Thanks for the report. Today the extension depends on the behavior of catkin_make in order to find the environment.

I'm sorry if I don't get this right, but … catkin tools are currently supported? And most of the time the ROS extension works well with catkin tools, so why was my issue changed to a request for a feature that already exists?

ooeygui, 2020-12-04:
@stertingen There are two sets of Catkin utilities - the official utilities like catkin_make and the *catkin tools*. We do not test the latter. We have had many reports of it not working. Since it isn't officially supported, this work item is about officially supporting it and testing.

FSund, 2021-03-16:
> @stertingen There are two sets of Catkin utilities - the official utilities like catkin_make and the _catkin tools_. We do not test the latter. We have had many reports of it not working. Since it isn't officially supported, this work item is about officially supporting it and testing.

From the [marketplace](https://marketplace.visualstudio.com/items?itemName=ms-iot.vscode-ros) page
> The build system (e.g. catkin_make or **catkin build**) will automatically be confirmed from the hidden files associated with each system.

Has this been implemented, or is the marketplace text referring to something ?

tkazik, 2021-05-21:
There seems to be a catkin-tools extension now, see [here](https://marketplace.visualstudio.com/items?itemName=betwo.b2-catkin-tools).

ooeygui, 2021-07-13:
Catkin tools is supported - https://github.com/ms-iot/vscode-ros/blob/master/src/build-tool/catkin-tools.ts. It looks like the check is currently not working.