ms-iot/vscode-ros #347: ROS2 Error: Launch request requires an absolute path as target

siferati, 2020-12-03:
I created a `launch.json` file as seen below:

```json
{
	"version": "0.2.0",
	"configurations": [
		{
			"name": "ROS: Launch",
			"type": "ros",
			"request": "launch",
			"target": "/workspaces/mypkg/install/mypkg/share/mypkg/launch/mylaunchfile.launch.py"
		}
	]
}
```

When I run it, I get the error `Launch request requires an absolute path as target` and nothing happens.

How can I launch my launch file?

seanyen, 2021-01-13:
@siferati Hi, thanks for reporting this issue. I am recently working on adding ROS2 launch debug support in this extension. And I'd like to invite you to try it from [our daily build](https://github.com/ms-iot/vscode-ros/suites/1811091640/artifacts/34801715) and see if it solves this issue. Thanks!

maciej-autobon, 2021-06-24:
@seanyen I just stumbled upon this Issue and your response and I was wondering: is `ros2 launch ...` command supported? Should I be able to enter a breakpoint with gdb when running my code this way?

ooeygui, 2021-06-24:
@maciej-autobon Yes, this should be supported. When running on Linux, we haven't been building with Symbols through the UI - that fix will be released shortly. If you pass `colcon build --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo` then breakpoints will be hit.

maciej-autobon, 2021-06-25:
@ooeygui Thanks for the answer! :)
I haven't tried it yet, all my launch commands are using `ros2 run ...` but I'll give the `ros2 launch ...` a go and let you know how it went.

maciej-autobon, 2021-06-26:
@ooeygui It works!
Awesome work you guys, thanks!

I've used [Allison Thackston template](https://www.allisonthackston.com/articles/vscode-docker-ros2.html) for ROS2 and vscode and I've pushed a [commit](https://github.com/MTDzi/ros2_vscode_test/commit/337cdc51f02e22b336d6c4231c66c2a459fcbbf1) (it's my other account) that tests out this `ros2 launch ...` functionality, in case anyone else is interested.