ms-iot/vscode-ros #353: [bug]

hfr2015, 2020-12-12:
(Please add appropriate labels)
- [ ] Windows: (Version)
- [x] Linux:  (Dist/Version)
ubuntu18.04
- [x] ROS 1: Dist 
Melodic
- [ ] ROS 2: Dist

\<Version of the plugin\>
v0.6.4
\<Copy the Version information from the `Help | About` menu\>
Version: 1.50.1
Commit: d2e414d9e4239a252d1ab117bd7067f125afd80a
Date: 2020-10-13T14:44:48.716Z
Electron: 9.2.1
Chrome: 83.0.4103.122
Node.js: 12.14.1
V8: 8.3.110.13-electron.0
OS: Linux x64 5.4.0-53-generic
# what is the bug
\<current behavior\>
I can't debug more than one rospy node in a launch file
I will get "Timed out waiting for launcher to connect" when I try to debug.
## Repro steps
\<how to reproduce the bug. Please attach sample files or github repo if appropriate.\>
1.launch file as flow:
<launch>

<param name="/global_des" value="heelo world" />

<group ns="test">
    <param name="local_info" value="in ns /test/my_test" />
    <param name="local_p" value="100" />
    <node pkg="local_planner_py" name="my_test_t2" type="parameter_server_test_clone.py" />
    <node pkg="local_planner_py" name="my_test" type="parameter_server_test.py" />
</group>
</launch>
2.
3.

# expected behavior
\<what you would expect to happen\>

# additional context
\<any additional information would be helpful for a better understanding\>

seanyen, 2021-01-13:
@hfr2015 Thanks for reporting. Do you mind to share all the versions of your installed extension? You can get it by `code --list-extensions --show-versions`.

ooeygui, 2021-06-08:
Closing; Please reopen with information if you can repro.

closed 2021-06-08 by ooeygui