ms-iot/vscode-ros #357: Extension causes high cpu load

shanky24-d, 2020-12-20:
- Issue Type: `Performance`
- Extension Name: `vscode-ros`
- Extension Version: `0.6.4`
- OS Version: `Windows_NT x64 10.0.19042`
- VSCode version: `1.52.1`

:warning: Make sure to **attach** this file from your *home*-directory:
:warning:`c:\Users\HP\AppData\Local\Temp\ms-iot.vscode-ros-unresponsive.cpuprofile.txt`

Find more details here: https://github.com/microsoft/vscode/wiki/Explain-extension-causes-high-cpu-load

ooeygui, 2020-12-22:
@shanky24-d  thank you for the report. Can you attach the cpu profile? (your comment only includes your local path, not the file itself)

shanky24-d, 2020-12-22:
@ooeygui 
cpu profile : [ms-iot.vscode-ros-unresponsive.cpuprofile.txt](https://github.com/ms-iot/vscode-ros/files/5729967/ms-iot.vscode-ros-unresponsive.cpuprofile.txt)

ooeygui, 2021-07-13:
https://code.visualstudio.com/api/working-with-extensions/bundling-extension.

ooeygui, 2021-11-07:
I'm working on the webpack for this, it should radically decrease load times and startup costs.

ooeygui, 2022-02-01:
This is a very significant change to the deployment of the VSCode ROS extension. I want to get some milage on it before releasing it.

ooeygui, 2022-10-13:
Resolving on as we have implemented webpack.

closed 2022-10-13 by ooeygui