ms-iot/vscode-ros #370: [feature] ROS extension to Visual Studio Community

bartholmberg, 2021-01-12:
ROS extension to Visual Studio Community

roycey, 2021-06-23:
@bartholmberg can you explain what you meant?

bartholmberg, 2021-06-23:
Visual Studio Code has/had ROS build extension [ e.g.- catkin].    Did not see this with VS Community.  I think the issue was there wasn't much available cross-platform for VS code with C++ [yet?]

ooeygui, 2021-06-23:
@bartholmberg The Visual Studio Code extension API isn't compatible with Visual Studio Community/Enterprise, To support Visual Studio, a completely new extension would need to be built. 
ASK: Which features of Visual Studio you'd like to use which aren't available in VSCode?

ooeygui, 2021-07-13:
I appreciate your interest in a Visual Studio ROS extension. Since this would be a new extension, I'm going to close this issue on the VSCode extension github. We will evaluate this for future investments.

closed 2021-07-13 by ooeygui