ms-iot/vscode-ros #374: [bug] ros2 cpp include dir

vincentrou, 2021-01-20:
- [ ] Linux:  ubuntu 20.04
- [ ]  ROS 2: foxy

plugin v0.6.6

vscode v1.52.1

# what is the bug

The file `.vscode/c_cpp_properties.json` does not contain underlayed workspace include path : `"/opt/ros/foxy/include/**"`

```json
{
  "configurations": [
    {
      "browse": {
        "databaseFilename": "",
        "limitSymbolsToIncludedHeaders": true
      },
      "includePath": [
        "/media/data/ws/colcon_ws/test_ws/src/examples/rclcpp/composition/minimal_composition/include/**",
        "/usr/include/**"
      ],
      "name": "ROS"
    }
  ],
  "version": 4
}
```
## Repro steps
1. F1 -> ros.updateCppProperties

# expected behavior
The file `.vscode/c_cpp_properties.json` contain underlayed workspace include path : `"/opt/ros/foxy/include/**"`

# additional context
ament env variable have the information :

`$ echo $AMENT_PREFIX_PATH`

`/media/data/ws/colcon_ws/test_ws/install/examples_rclpy_minimal_subscriber:/media/data/ws/colcon_ws/test_ws/install/examples_rclpy_minimal_service:/media/data/ws/colcon_ws/test_ws/install/examples_rclpy_minimal_publisher:/media/data/ws/colcon_ws/test_ws/install/examples_rclpy_minimal_client:/media/data/ws/colcon_ws/test_ws/install/examples_rclpy_minimal_action_server:/media/data/ws/colcon_ws/test_ws/install/examples_rclpy_minimal_action_client:/media/data/ws/colcon_ws/test_ws/install/examples_rclpy_executors:/media/data/ws/colcon_ws/test_ws/install/examples_rclcpp_multithreaded_executor:/media/data/ws/colcon_ws/test_ws/install/examples_rclcpp_minimal_timer:/media/data/ws/colcon_ws/test_ws/install/examples_rclcpp_minimal_subscriber:/media/data/ws/colcon_ws/test_ws/install/examples_rclcpp_minimal_service:/media/data/ws/colcon_ws/test_ws/install/examples_rclcpp_minimal_publisher:/media/data/ws/colcon_ws/test_ws/install/examples_rclcpp_minimal_composition:/media/data/ws/colcon_ws/test_ws/install/examples_rclcpp_minimal_client:/media/data/ws/colcon_ws/test_ws/install/examples_rclcpp_minimal_action_server:/media/data/ws/colcon_ws/test_ws/install/examples_rclcpp_minimal_action_client:/opt/ros/foxy`

agutenkunst, 2021-04-02:
Seems like with ros2 the extension is trying to get the include path from the environment variable `COLCON_PREFIX_PATH` see 

https://github.com/ms-iot/vscode-ros/blob/a50f699259e9e397e7d05c7c6e429d8bb8c2ea94/src/ros/ros2/ros2.ts#L64-L80

but maybe it should use `AMENT_PREFIX_PATH`. But I am not sure what workspace setups exist in which distro. Also see https://github.com/ros2/ros2/issues/855#issuecomment-575862776

But maybe @dirk-thomas can confirm that `AMENT_PREFIX_PATH` is the better choice.

dirk-thomas, 2021-04-08:
(I am sorry, but I am not involved with the project at this point in time. So my answer might be very brief.)

`COLCON_PREFIX_PATH` isn't available when packages are build without `colcon`. That is e.g. the case for the Debian packages. Therefore I think `AMENT_PREFIX_PATH` would be a better choice.

ooeygui, 2021-04-18:
@dirk-thomas Thank you for the feedback - @agutenkunst Thank you for the contribution!

PR #423 (2021-04-18): Use AMENT_PREFIX_PATH instead of COLCON_PREFIX_PATH. Closes #374
Determine the includeDirs to set in .vscode/c_cpp_properties.json
from AMENT_PREFIX_PATH.

I checked with eloquent and foxy and it worked for both.

Closes #374
commit 901a8103 (2021-04-18): Use AMENT_PREFIX_PATH instead of COLCON_PREFIX_PATH. Closes #374 (#423)

Determine the includeDirs to set in .vscode/c_cpp_properties.json
from AMENT_PREFIX_PATH
closed 2021-04-18 by ooeygui